mcci-catena/arduino-lmic #629: EV_TXCOMPLETE, no JoinAccept or waiting for RX windows

ptkuo0322, 2020-11-23:
- Version of LMIC being used:  3.2.0
- Version of Arduino IDE being used: 1.8.13
- Network provider The Things Network
- Region US915
- Board Arduino Uno
- Radio HopeRF RFM95W (Sx1276)

Here is my problem. I tried to use the example code "ttn-otaa" to help me understand how the LoRaWAN works. 
I double check the 
 1. project_config to include the   CFG_us915 1 and  CFG_sx1276_radio 1
 2. I add the code LMIC_setClockError(MAX_CLOCK_ERROR * 10 / 100); LMIC_setLinkCheckMode(1); LMIC_setAdrMode(1); 
     LMIC.dn2Dr = DR_SF10;LMIC_setDrTxpow(DR_SF7, 14) in the setup right after the LMIC_reset.
 3. I also double check the APPEUI(lsb), DECEUI(lsb) and APPKET(msb) to make sure that they are set up correctly.

However, for the information that print in the serial monitor, I would get the following 
02:24:25.103 -> 12737330: EV_JOIN_TXCOMPLETE: no JoinAccept
02:24:39.168 -> 13615511: EV_TXSTART
02:24:44.642 -> 13956081: EV_JOINED
02:24:44.676 -> netid: 19
02:24:44.676 -> devaddr: 26012A7A
02:24:44.710 -> AppSKey: 21-DB-FA-7D-37-86-E1-3F-D4-1F-89-4D-ED-B5-94-0A
02:24:44.779 -> NwkSKey: CF-55-AB-8A-4D-C9-90-7E-C0-52-3C-98-D0-3C-BE-13
02:24:44.813 -> 13962753: EV_TXCOMPLETE (includes waiting for RX windows)
02:25:44.997 -> 17717009: EV_TXSTART
02:25:44.997 -> Packet queued
02:25:47.361 -> 17864463: EV_TXCOMPLETE (includes waiting for RX windows)
02:26:47.524 -> 21614955: EV_TXSTART
02:26:47.559 -> Packet queued
02:26:49.892 -> 21762410: EV_TXCOMPLETE (includes waiting for RX windows)
02:27:50.067 -> 25512903: EV_TXSTART

I do not know what is the problem and I already searched for many reference on the internet. But I still do not know how to fix this problem for my first test. Can anyone give me some hints? I would really appreciate it.

I only change the following three parts in the example code.

`static const u1_t PROGMEM APPEUI[8]={ 0x3B, 0x89, 0x03, 0xD0, 0x7E, 0xD5, 0xB3, 0x70 };
void os_getArtEui (u1_t* buf) { memcpy_P(buf, APPEUI, 8);}

// This should also be in little endian format, see above.
static const u1_t PROGMEM DEVEUI[8]={ 0x63, 0x75, 0xBC, 0xEB, 0x16, 0x94, 0x16, 0x95 };
void os_getDevEui (u1_t* buf) { memcpy_P(buf, DEVEUI, 8);}

// This key should be in big endian format (or, since it is not really a
// number but a block of memory, endianness does not really apply). In
// practice, a key taken from ttnctl can be copied as-is.
static const u1_t PROGMEM APPKEY[16] = { 0xD4, 0xD2, 0xDD, 0x70, 0x7E, 0xE0, 0xD2, 0x01, 0xDF, 0xD3, 0x92, 0xEA, 0x10, 0x87, 0xA5, 0x3F };
void os_getDevKey (u1_t* buf) {  memcpy_P(buf, APPKEY, 16);}

`// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 9,
    .dio = {2, 3, LMIC_UNUSED_PIN},
};`

`static const u1_t PROGMEM APPEUI[8]={ 0x3B, 0x89, 0x03, 0xD0, 0x7E, 0xD5, 0xB3, 0x70 };
void os_getArtEui (u1_t* buf) { memcpy_P(buf, APPEUI, 8);}

// This should also be in little endian format, see above.
static const u1_t PROGMEM DEVEUI[8]={ 0x63, 0x75, 0xBC, 0xEB, 0x16, 0x94, 0x16, 0x95 };
void os_getDevEui (u1_t* buf) { memcpy_P(buf, DEVEUI, 8);}

// This key should be in big endian format (or, since it is not really a
// number but a block of memory, endianness does not really apply). In
// practice, a key taken from ttnctl can be copied as-is.
static const u1_t PROGMEM APPKEY[16] = { 0xD4, 0xD2, 0xDD, 0x70, 0x7E, 0xE0, 0xD2, 0x01, 0xDF, 0xD3, 0x92, 0xEA, 0x10, 0x87, 0xA5, 0x3F };
void os_getDevKey (u1_t* buf) {  memcpy_P(buf, APPKEY, 16);}
`

``

terrillmoore, 2020-11-24:
Hi, sorry you're having problems.

Since join works, we know that downlink is working.  What do you see in the TTN console?

--Terry

ptkuo0322, 2020-11-24:
> Hi, sorry you're having problems.
> 
> Since join works, we know that downlink is working. What do you see in the TTN console?
> 
> --Terry

Hi, this is what I have seen on the TTN
[https://imgur.com/Gg18MEF](url)

and I notice that the RSSi is pretty weak, it around -120. is that a problem?

ptkuo0322, 2020-11-25:
> Hi, sorry you're having problems.
> 
> Since join works, we know that downlink is working. What do you see in the TTN console?
> 
> --Terry

Later today, I did not change anything, but I always got  \Documents\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src\lmic\oslmic.c:53
in my serial monitor...

I have no idea what happen.. I check the pin for more than 10 times to make sure that they match with pin map. And the library version is 0.8.0 and I use arduino uno and hopeRF RFM95W Lora.

Could somebody give me a hint

terrillmoore, 2020-11-26:
That means your HopeRF is not connected properly on SPI to your Arduino.

ptkuo0322, 2020-12-03:
> That means your HopeRF is not connected properly on SPI to your Arduino.

It is really weird. I hook the Lora to 3.3V a few days ago, and it works fine. However, a few days later, I need to hook it to 5V with the same wires, Arduino uno board and Lora module.

cyberman54, 2020-12-03:
@ptkuo0322 The RFM95 is for 3,3V only. If you connected it to 5V, you may have damaged it.

[image]

terrillmoore, 2020-12-06:
Might be better to discuss this on [forum.mcci.io](https://forum.mcci.io) -- this doesn't look like a bug per se.

closed 2020-12-10 by terrillmoore