mcci-catena/arduino-lmic #651: How to create dynamically-selectable Lorawan region / channel selection at startup

pomplesiegel, 2020-12-17:
For a device which could be used in many regions across the globe (assuming a corresponding variant of modem + antenna is installed), what is the best way to programmatically select a **pre-defined channel plan per region** at startup? 

The goal is to avoid having unique binaries for each possible channel plan (adds up very quickly once you consider all possible global regions). I see how to make this change in code, but this would produce unique binaries for each region.

````cpp
#define CFG_us915 1
//#define CFG_au915 1
````

This is what I imagine for a workflow 
- Device starts up and reads from dipswitch / jumper configuration values to indicate what region it is in and what modem is installed
- Device programmatically loads appropriate channel plan into LMIC and begins normal LoraWAN join behavior

Any thoughts? Forgive me if this is an obvious question.

Thank you,
Michael

pomplesiegel, 2021-01-22:
@terrillmoore, Any idea if this could be achieved within LMIC, or is this just way too far out of scope, given the amount of region-dependent `#define`s, behaviors and symbols?

terrillmoore, 2021-01-23:
It can definitely be done. It will make the code footprint probably somewhat larger but maybe not by much. Here's what has to be done.

1. Refactor lmic.h so that, instead of #if uslike/eulike, you have a union inside struct lmic with two variants: uslike and eulike.
2. Throughout the code, replace the current function calls to region-specific handlers with calls to a dispatcher that dispatches based on the region -- follow the pattern in the secure element branch for dispatching to secure elements. There will be some refactoring.
3. Figure out how to let people get back to the same size as today by "clever" #ifdefs in the shim module.
4. (Important but not strictly part of the above): provide APIs for saving/restoring LMIC state across reboots.
5. Update examples (which needs to be done anyway).

Take about a week, I reckon (with test). Problem is finding the week!

This would break some user code, because the names of fields would change. Some legacy #defines could be provided to ease porting, but this has to be done with care because of name-space pollution. My preference would be to require user code to either `#include "lmic_legacy_v3.h"` or change source code to adapt to the new reality.

pomplesiegel, 2021-01-23:
> Take about a week, I reckon (with test). Problem is finding the week!
Of course, especially these days! Well, this is still very exciting to know it can be done