mcci-catena/arduino-lmic #652: ttn-otaa example compiling error Ardino ide

dragosmacovei, 2020-12-27:
I tried to compile the example `ttn-otaa.ino`. The first step I tried was to 

`#define COMPILE_REGRESSION_TEST 1`

In order to run the test. I got the following compilation errors:

```
In function 'void onEvent(ev_t)':
ttn-otaa:119:67: error: 'LMIC_getSessionKeys' was not declared in this scope
               LMIC_getSessionKeys(&netid, &devaddr, nwkKey, artKey);
                                                                   ^
ttn-otaa:194:14: error: 'EV_TXSTART' was not declared in this scope
         case EV_TXSTART:
              ^
ttn-otaa:197:14: error: 'EV_TXCANCELED' was not declared in this scope
         case EV_TXCANCELED:
              ^
ttn-otaa:200:14: error: 'EV_RXSTART' was not declared in this scope
         case EV_RXSTART:
              ^
ttn-otaa:203:14: error: 'EV_JOIN_TXCOMPLETE' was not declared in this scope
         case EV_JOIN_TXCOMPLETE:
              ^
exit status 1
'LMIC_getSessionKeys' was not declared in this scop
```

- Last LMIC version downloaded just today
- Network provider (The Things Network)
- Region (EU868)
- Radio (SX1276 - **I believe, not sure how to find out**)
- Arduino: 1.8.14 Hourly Build 2020/12/15
- Os: (Windows 10)
- Board: "TTGO T1" (TTGO T-Beam v1.1) Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600

dragosmacovei, 2020-12-27:
The problem was solved downgrading to Arduino 1.8.13

closed 2020-12-27 by dragosmacovei

Erhannis, 2024-11-13:
I had a similar problem, but mine appeared to be because I had an "LMIC-Arduino" library installed, before "MCCI_LoRaWAN_LMIC_library".  Disabling the prior library fixed that problem.  (Though now I'm getting a "multiple definition of 'hal_init'" error I still have to debug.)