mcci-catena/arduino-lmic #659: Confusion regarding Asia channel plans

pomplesiegel, 2021-01-11:
Hello Everyone,

I am a bit confused regarding the Asia channel plans. On [TTN's site](https://www.thethingsnetwork.org/docs/lorawan/frequencies-by-country.html), different regions use either AS920-923 ("AS1") or AS923-925 ("AS2"). 

[image]

[image]

However, in LMIC I see only two possible "Asia" channel plan options 

````cpp
//#define CFG_as923 1
// #define LMIC_COUNTRY_CODE LMIC_COUNTRY_CODE_JP	/* for as923-JP */
````

What selection should someone use if they were to want to deploy in an area like Singapore, versus Taiwan? 

Thank you,
Michael

terrillmoore, 2021-01-11:
Devices always use AS923 or AS923_JP.  This is because the JOIN acknowledge tells you what channels to use. (But don't tell you about LBT requirements). So: device firmware doesn't know about AS923 AS1 vs AS2; that's only relevant to gateways. In fact, the regional plan doesn't talk about the channels the gateways use; it just turns out that there are two ways.

Of course, if you're doing ABP you have to load your channels in your firmware, and then you must (as in all things) ask your network operator what to do.

Aside: I feel that ABP is really only suitable for advanced users -- don't use this because it seems simpler, it really isn't. It's much more complex, because it involves you in advanced aspects of the network. Much better to use OTAA and let the network tell the LMIC what to do.

pomplesiegel, 2021-01-11:
Got it, thank you! Yes, I'm talking about OTAA. This info is very helpful. Closing this "issue" :)

closed 2021-01-11 by pomplesiegel