mcci-catena/arduino-lmic #664: Maduino Lora Radio (915M)

iaomari93, 2021-01-18:
hello,
Im trying to use Maduino Lora Radio [(915M)](https://www.makerfabs.com/maduino-lora-radio-433m-868m-915m.html) with chripStack and mikrotik WAP LoRa9 gateway ,us 915 sub1 and I’m facing a few problems with it.

I’m trying the ttn_abp example from MCCI lorawan lmic lib v3.2.0. I can see the Unconfirmed Data Up and Unconfirmed Data Down in the gateway traffic page.
[image]

from the arduino serial monitor Im getting :
Starting
9654: EV_TXSTART
Packet queued
139471: EV_TXCOMPLETE (includes waiting for RX windows)
3890090: EV_TXSTART
Packet queued
4019924: EV_TXCOMPLETE (includes waiting for RX windows)
…

I tried both of thus pin mapping:
// Pin mapping
const lmic_pinmap lmic_pins = {
.nss = 10,
.rxtx = LMIC_UNUSED_PIN,
.rst = 9,
.dio = {2, 6, LMIC_UNUSED_PIN},
};

and :
// Pin mapping
const lmic_pinmap lmic_pins = {
.nss = 10,
.rxtx = LMIC_UNUSED_PIN,
.rst = 9,
.dio = {2, 6, 7},
};

and for both of them I’m not getting the message back from the gateway.
I have been trying for a long time. I really appreciate if anyone know what could the problem be.

iaomari93, 2021-01-18:
it still not working,
now I'm trying otaa example code with the same setup.
in the gateway side I can see pairs of join-request and join-accept.
but from the serial monitor is stuck in this loop

Starting
Packet queued
2332: EV_JOINING
2892: EV_TXSTART
408007: EV_JOIN_TXCOMPLETE: no JoinAccept
433054: EV_TXSTART
816867: EV_JOIN_TXCOMPLETE: no JoinAccept
826228: EV_TXSTART
1231352: EV_JOIN_TXCOMPLETE: no JoinAccept
1265829: EV_TXSTART
1649646: EV_JOIN_TXCOMPLETE: no JoinAccept
1720815: EV_TXSTART
2125945: EV_JOIN_TXCOMPLETE: no JoinAccept
2184518: EV_TXSTART
2568336: EV_JOIN_TXCOMPLETE: no JoinAccept
3192361: EV_TXSTART
3597491: EV_JOIN_TXCOMPLETE: no JoinAccept
3603107: EV_TXSTART
3986924: EV_JOIN_TXCOMPLETE: no JoinAccept
4566180: EV_TXSTART
4971310: EV_JOIN_TXCOMPLETE: no JoinAccept
5003058: EV_TXSTART
5386875: EV_JOIN_TXCOMPLETE: no JoinAccept
6121246: EV_TXSTART
6526376: EV_JOIN_TXCOMPLETE: no JoinAccept
6530296: EV_TXSTART
6914115: EV_JOIN_TXCOMPLETE: no JoinAccept
7448166: EV_TXSTART
7853297: EV_JOIN_TXCOMPLETE: no JoinAccept
7855732: EV_TXSTART
8239552: EV_JOIN_TXCOMPLETE: no JoinAccept
9105023: EV_TXSTART
9510155: EV_JOIN_TXCOMPLETE: no JoinAccept
9544384: EV_TXSTART
9928202: EV_JOIN_TXCOMPLETE: no JoinAccept
**9928306: EV_JOIN_FAILED**
10777321: EV_TXSTART
11182459: EV_JOIN_TXCOMPLETE: no JoinAccept
11185006: EV_TXSTART
.
.
.

thank you
omari

terrillmoore, 2021-02-20:
I suggest you first use two boards with the raw example. Get that working and confirm receive operation.

Then use the compliance sketch (without any application logic) and see if able to join and send messages.

Downlink problems are always timing related.

Also, current version is 3.3. Is there a reason not to use that?

Best regards,
--Terry

joecryptotoo, 2021-04-10:
@iaomari93  Did you ever get this working? I'm having the same issues.

const lmic_pinmap lmic_pins = {
.nss = 10,
.rxtx = LMIC_UNUSED_PIN,
.rst = 9,
.dio = {2, 6, 7},
};

This is configuration is right and it works with raw.

iaomari93, 2021-04-10:
@terrillmoore, thank you for the suggestion. it was indeed a timing related issue.

@joecryptotoo, you are right, the configuration is right :). my problem was with the configuration file of the NS that I used. what NS you are using? what frequency plan? gateway?

paulcastor30, 2021-04-16:
@iaomari93 have you got this problem solved? i am facing the same issue.

terrillmoore, 2021-04-16:
All: 

1. ChirpStack is substantially configurable, and when I tested, I found it was not right "out of the box" for 915. I had no big problem because I had lots of experience with TTN and the LMIC, but I think it is not transparent if you're just getting started. Please test with a network that you *don't* have to configure, like TTN, before testing with ChirpStack.

2. ABP seems simpler, but in fact is *also* and advanced option.

To bring up, please:  

1. Test OTAA with TTN and make sure it works (and that you understand what's going on)
2. Then try OTAA with ChirpStack
3. Only then, try ABP (with TTN first, please!)
4. Then finally try ABP with ChirpStack

There are too many knobs on ChirpStack; unless you have verified hardware and some network experience, it's not the right place to start. Any gateway that works with ChirpStack can trivially work with TTN: so use that first, then convert.