mcci-catena/arduino-lmic #666: EV_JOIN_TXCOMPLETE: no JoinAccept with TTGO-LoRa 2.1.6

AlexandreMouyen, 2021-01-27:
Hello there,

I am currently using a TTGO LoRa 2.16 in order to connect to the The Things Network. I replaced all the fields on the example sketch (ttn-otaa) from my TTN console. It has already been two days that I try, I every time have the "EV_JOIN_TXCOMPLETE: no JoinAccept" message. 

There are my pins configuration:
`
const lmic_pinmap lmic_pins = {
    .nss = 18,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 14,
    .dio = {26, 33, 32},
};
`

I am also using the CFG_eu868 config.

Can someone help me? Did I miss something?

haflinger, 2021-02-19:
Same issue here. 

 I use : ```const lmic_pinmap lmic_pins = {
    .nss = 18, 
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 23,
    .dio = {/*dio0*/ 26, /*dio1*/ 33, /*dio2*/ 32},
};```

It works with ABP activation method but with OTAA i have `EV_JOIN_TXCOMPLETE: no JoinAccept`

terrillmoore, 2021-02-20:
Hi @haflinger, we cannot use debug prints in the LMIC. Seriously. It disturbs timing to the point that you can't do downlinks. I have not taken them out becaue they're useful for debuggigng logic. But they are useless if the LMIC is not running on your platform because of timing events. This is not (even in theory) possible to fix, unless your debug print is infinitely fast and never blocks. The formatting from printf is *not* infinitely fast, and prints on Arduino always are blocking.

This is why I added the debug log functionality. I'm familiar with this from other protocols

I left debug prints in the code because I wanted to minimize the deviation from the original; but they're not useful unless the LMIC is crashing. Please turn them off when trying to debug downlink problems.

Thanks,
--Terry

arangu97, 2021-03-02:
Hello, I am also using that board with the OTAA example and the following lines are printed in the Serial Monitor: 

Packet queued
4804: EV_JOINING
96155: Unknown event
17

The Activation message appears in TTN : 
[image]

However it seem like the LoRa device is not getting the JOIN ACCEPT, an unknown event instead.

What could be happening? 

Thank you

robputt796, 2021-04-08:
Hello,

I am using a TTGO T-Beam, so similar setup ESP32 and SX1276. I am also having issues with OTAA you can see my debug output here - https://www.thethingsnetwork.org/forum/t/otaa-v3-acting-weird/46394/6

Best Regards,

Rob

cyberman54, 2021-04-12:
Are you using full LoRaWAN compliant 8ch gateways? Otherwise OTA can't work.
See https://github.com/mcci-catena/arduino-lmic/issues/689

robputt796, 2021-04-13:
@cyberman54 - Sure am