mcci-catena/arduino-lmic #670: NB_Rep - Strange behaviour

Cloolalang, 2021-02-06:
mcci-catena/arduino-lmic 3.2.0  on TTGO v1 EU868, Rak Gateway with internal Loraserver.

I would like to be able to change/set/control NB_Rep ( then number of times uplink packets are transmitted (with the same frame number) as this is a redundancy feature for packet-loss control in the LoRaWan specification (amongst other parameter such as TX_Power and SF).

So I dug around in this library, and found in lmic.c, line 602:

LMIC.upRepeat = 0;

I found that if I change this I do start seeing repeating uplinks with same frame number, great it works!

BUT:

When I changed it back my TTGO still sends repeated uplinks!, oh no!!
(ADR is set to OFF)

I did something to permanently set my node to send repeated uplinks and the only why I can stop that now is to set
LMIC.upRepeat = 0;
before every transmission (LMIC_setTxData2…)

Any Idea how I manged to do this, or how I can fix it?

Thanks very much for this great project, its been very useful to be able to tweek the LoraWan parameters

Cheers
P

terrillmoore, 2021-02-06:
Well, I'll spare you a long discussion; there's no proper API for using this. However, i would experiment with resetting `LMIC.upRepeatCount` whenever you change `LMIC.upRepeat`.  Other than that, no idea why you're getting repeated uplinks when `LMIC.upRepeat` is zero.

Cloolalang, 2021-02-07:
Thanks Terry, I'll take a look an d let your know what I discover!
I really appreciate your insights and all that you have done so far with LMIC
Cheers
Patrick

closed 2021-02-07 by Cloolalang