mcci-catena/arduino-lmic #689: Problem with OTAA activation with TTN

arangu97, 2021-03-04:
Hello, I am using the TTGO Esp32 with SX1276 LoRa module. 

I am using the last version of LMIC and Arduino 1.8.12 version.

The problem is that ABP is working fine with TTN, but however OTAA is not working properly. 

The following is appearing with Debug verbose mode. I have also tried to disable the debug mode and just : EV_JOINING >> EV_TXSTART >> EV_JOIN_TXCOMPLETE: no JoinAccept appear.

> RXMODE_RSSI
> 4736: engineUpdate, opmode=0x808
> Packet queued
> 4759: EV_JOINING
> 4764: engineUpdate, opmode=0xc
> 400279: engineUpdate, opmode=0xc
> 400296: EV_TXSTART
> 400364: TXMODE, freq=868500000, len=23, SF=7, BW=125, CR=4/5, IH=0
> 714730: setupRx1 txrxFlags 00 --> 01
> start single rx: now-rxtime: 4
> 715362: RXMODE_SINGLE, freq=868500000, SF=7, BW=125, CR=4/5, IH=0
> rxtimeout: entry: 718386 rxtime: 715355 entry-rxtime: 3031 now-entry: 4 rxtime-txend: 311126
> 776980: setupRx2 txrxFlags 0x1 --> 02
> start single rx: now-rxtime: 4
> 777612: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
> rxtimeout: entry: 794012 rxtime: 777605 entry-rxtime: 16407 now-entry: 4 rxtime-txend: 373376
> 794029: processRx2Jacc txrxFlags 0x2 --> 00
> 794083: EV_JOIN_TXCOMPLETE: no JoinAccept
> 794317: engineUpdate, opmode=0xc

I can see the JoinRequest and JoinAccept in TTN and also in my Dragino LG02 LoRa gateway(as I attach in the images). 
It seems like the downlink from Dragino to the TTGO Esp32 is not working.

Thank you. 

[image]
[image]

cyberman54, 2021-04-12:
@arangu97 Dragino LG02 is a "dual channel" LoRa Gateway, what means, it is NOT a LoRaWAN compliant gateway. So this can't work with LMIC. You need a compliant 8 channel gateway.

This is not an MCCI LMIC realated issue -> should be closed.

arangu97, 2021-04-13:
Ok, I understand... 

So ABP is working because no dowlink is needed from the gateway but OTAA procedure is not getting done properly. That is right?

cyberman54, 2021-04-13:
For OTAA you need downlink, for ABP not. But the network needs also downlinks to control the nodes by MAC commands, so it's not a good idea to operate nodes without downlink.

LORAWAN networks use channel plans to hop between channels, e.g. 8 channels in EU868 band.

If your gateway has poor radio hardware which supports only 1 or 2 channels, it will miss 6 or 8 channels of the network server. So there is a certain chance that OTAA joins go through, but this is non deterministic.

arangu97, 2021-04-13:
Ok thank you so much. 
Do you recommend any 8 channel LoRa gateway to work with TTN ?

cyberman54, 2021-04-13:
Please, close this issue here now and use forums, e.g. MCCI forum or TTN forum. Thanks.

closed 2021-04-13 by arangu97