mcci-catena/arduino-lmic #693: Starting Failure Problem Arduino Nano + sx1278 (RA-02)

paulcastor30, 2021-03-28:
**Consider raising support questions on the forum first**

There is a discussion site, [forum.mcci.io](https://forum.mcci.io), which includes a [category focused on the Arduino LMIC](https://forum.mcci.io/c/device-software/arduino-lmic/5). Unless you're sure that your problem is a bug in the LMIC, it would be great if you can try for help there first. That will keep issues focused on work that needs to be done by the developers.

**Describe your question or issue**

Please give a clear and concise description of the problem you're facing and what you'd like help with.

**Environment**

This information is very important; it's hard to help without a complete set of answers.

- Version of LMIC being used. If using the latest github mainline, let us know, otherwise state the version.
- Version of Arduino IDE being used:
- Network provider (The Things Network, Swisscom, ChirpStack, etc.)
- Region (EU868, US915, etc.)
- Board (MCCI Catena, Adafruit Feather M0, Heltec Wi-Fi LoRa 32 v2, etc.)
- Radio (HopeRF, SX1276, etc.)

I Followed the guide and successfully uploaded the problem but the respod in the serial monitor of my arduino is like this:
Starting
FAILURE 
C:\Users\Paul Rodolf Castor\Documents\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src\lmic\radio.c:923

Can i ask a help to clear this problem to get started with LoRaWAN in TTN?

IoTThinks, 2021-03-29:
Try to use external power for LoRa.

Arduino Nano is likely to provide insufficient power to LoRa.

paulcastor30, 2021-03-29:
Thank you very much for you great comment. i changed my arduino nano to arduino mega and it worked.

closed 2021-03-29 by paulcastor30