mcci-catena/arduino-lmic #695: No EV_TXCOMPLETE event

tbornon, 2021-03-31:
**Describe the bug**

No EV_TXCOMPLETE event is fired.

**Environment**

- mcci-catena/MCCI LoRaWAN LMIC library@^3.3.0
- Platformio with VSCode (Core 5.1.1, Home 3.3.4)
- TTN v3
- EU868
- Adafruit Feather 32u4 LoRa
- ATMega 32u4
- RFM95C
- Build flags : 
        -D ARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS
	-D CFG_eu868=1
	-D CFG_sx1276_radio=1
	-D DISABLE_PING
	-D DISABLE_BEACONS
	-D LMIC_LORAWAN_SPEC_VERSION=LMIC_LORAWAN_SPEC_VERSION_1_0_3

**To Reproduce**

Code can be found there : https://gist.github.com/tbornon/05a4429068546dba98ac0673ab69b4f3

**Expected behavior**

The EV_TXCOMPLETE should happen after a EV_TXSTART

**Screenshots**

[image]

terrillmoore, 2021-04-07:
TTN V3? You have to wait 5 seconds for TX complete. See discussion at https://github.com/mcci-catena/arduino-lmic/discussions/699.

tbornon, 2021-04-07:
Hi @terrillmoore , 
Thanks for your answer. I've effectivly seen that TTN v3 is using a 5s RX delay but as you can see, my TXSTART is at 18,6ms and therefore the TXCOMPLETE should happend at around 23,6ms but it doesn't.
Each time division is 5s so it means I've waited until 27ms without TXCOMPLETE event. It should have covered RX1 and RX2 windows. I even tried to wait for a few minutes with much more success.

tbornon, 2021-04-08:
I ended up fixing the problem. It was a stupid mistake, I copied the pin mapping for the Feather 32u4 from the exemples provided without checking it.
By default, Feather 32u4's pin 6 isn't connected to DIO1. 

I proposed PR #703 to add a warning for the users

closed 2021-04-08 by tbornon