mcci-catena/arduino-lmic #697: Title in README.md should read MCCI LoRaWAN LMIC library

lnlp, 2021-04-06:
**What document is involved?**

[README.md](https://github.com/mcci-catena/arduino-lmic/blob/master/README.md)

**What changes are needed?**

The title should be changed to the official library name: "MCCI LoRaWAN LMIC library"

The current title reads "Arduino-LMIC library" but this neither reflects the library name nor correctly reflects the repository name. 

The title was inherited from the repository's predecessor where it lacked the same issue (neither reflect the library name nor correctly reflect the repository name).

terrillmoore, 2021-04-06:
Good catch, thanks, I'll update.

lnlp, 2021-04-06:
> Good catch

It took me some years.. 🤣

PR #748 (2021-06-01): Bug fixes and doc improvements
Fix #697 (docs)

Fix #677 (don't transmit when busy)

Other minor improvements
commit b04fb14f (2021-06-01): Fix #697: clarify title vs library name in IDE
closed 2021-06-01 by terrillmoore