mcci-catena/arduino-lmic #706: E (133) psram: PSRAM ID read error: 0xffffffff

manivalaguru, 2021-04-12:
hello,
   i am using esp32 with nice rf modules to act as a node i have already created my gateway i have changed the region in configuration and changed the pins according to my board but when i compile following error ??can you help me resolve the issue
138871: EV_TXCOMPLETE (includes waiting for RX windows)
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
E (133) psram: PSRAM ID read error: 0xffffffff
Starting
8142: EV_TXSTART
Packet queued
138785: EV_TXCOMPLETE (includes waiting for RX windows)
3888815: EV_TXSTART
Packet queued
4019448: EV_TXCOMPLETE (includes waiting for RX windows)
7769479: EV_TXSTART
Packet queued
7900112: EV_TXCOMPLETE (includes waiting for RX windows)
11650142: EV_TXSTART
Packet queued
11780775: EV_TXCOMPLETE (includes waiting for RX windows

**Environment**

This information is very important; it's hard to help without a complete set of answers.

- Version of LMIC being used. If using the latest github mainline, let us know, otherwise state the version.
- Version of Arduino IDE being used:
- Network provider (The Things Network, Swisscom, ChirpStack, etc.)
- Region (EU868, US915, etc.)
- Board (MCCI Catena, Adafruit Feather M0, Heltec Wi-Fi LoRa 32 v2, etc.)
- Radio (HopeRF, SX1276, etc.)

cyberman54, 2021-04-12:
This is NOT an LMIC related issue.

It's caused by your Arduino IDE setting which probably don't match your ESP32 hardware, i.e. you need a PSRAM chip on your board.

Should be closed here.

terrillmoore, 2021-04-15:
I agree about the PSRAM error. Not clear from info whether fixing the PSRAM problem will fix the LMIC. @manivalaguru please fix that, and then if there are problems update this ticket (and edit the title).

terrillmoore, 2021-04-18:
Closing this for now. Reopen if there's still a problem.

closed 2021-04-18 by terrillmoore