mcci-catena/arduino-lmic #707: Support for SX1301 Target Radio

Faizan2253, 2021-04-12:
Hi there,
I have LoRa gateway (RAK831) which has sx1301 chip for LoRa connectivity. My nodes are based on sx1278 chip. Frequency band is 433Mhz. In selecting the target radio transceiver sx1301 is not defined (Since this is my target radio). Will  this library still work on my nodes with the hardware that I have or am I misunderstood something? (The Link of my gateway details are attached).
Thanks!!!!

https://docs.rakwireless.com/Product-Categories/WisLink/RAK831/Datasheet/#board-overview

terrillmoore, 2021-04-12:
Hi @Faizan2253,

Thanks for your question. The LMIC does not have support for the 433 MHz bands. It could; I believe all the logic is there, and I don't think the sx1278 would be an issue; but we've never done a regional plan for 433 MHz, so it's not been tested at all, and there's work to be done. Which regional plan are you using?

Best regards,
--Terry

Faizan2253, 2021-04-12:
@terrillmoore  Thanks for your reply sir.
I am bit confuse about the term target radio that is used in documentation which can be selected in config. Is it referred to radio on node or on Gateway?
And secondly is it possible to . make changes in this library to make it compeitble with 433Mhz hardware.
I am from Pakistan. According to LoRa Alliance documentation following regions are supported. But my hardware is for 433 Mhz.
[image]

terrillmoore, 2021-04-12:
Target radio refers to the radio on the edge device, not on the gateway.

The LMIC currently is lacking support for EU433. It could be added, but it would have to be a community contribution as I have no time to look into adding it over the next few months.

Faizan2253, 2021-04-13:
@terrillmoore Thanks for your guidance Sir.