mcci-catena/arduino-lmic #708: Support request for UNO r3 + SX1272MB2xAS on 868 freq

maitrekobayashi, 2021-04-15:
Hi, 
I'm new to embedded programmation and I'm currently facing an issue on my UNO R3 + SX1272 configuration. (objenious network, 868MHz, France)

I set the right parameters on lmic_project_config.h :
**#define CFG_eu868 1
#define CFG_sx1272_radio 1**

The console returns :
**Starting
FAILURE 
C:\...\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src\lmic\oslmic.c:53**

which lead to this function : 

**void os_init() {
    if (os_init_ex((const void *)&lmic_pins))
        return;
    ASSERT(0);
}**

I'm working with Arduino IDE 1.8.13, and i add #define DISABLE_PING in the main.

Any idea of what the problem could be ? Maybe the UNO R3 do not fit ?
Thanks

terrillmoore, 2021-04-15:
You'd get a failure other places if it didn't fit. You should add DISABLE_PING (and DISABLE_BEACONS) in your lmic_project_config.h; but this failure is due to `os_init_ex()` not liking your lmic_pins object, and that's much lower level. Can you look at the code for os_init_ex() and find the paths that will return false, and add ASSERT(0) in each place until you track it down?

maitrekobayashi, 2021-04-19:
Hi, thanks for answering, 

I tried to track down the error, and it led me to C:\...\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src\lmic\radio.c:1164
which is : 

_u1_t radio_rand1 () {
    u1_t i = randbuf[0];
    ASSERT( i != 0 );
    if( i==16 ) {
        os_aes(AES_ENC, randbuf, 16); // encrypt seed with any key
        i = 0;
    }
    u1_t v = randbuf[i++];
    randbuf[0] = i;
    return v;
ASSERT(0);
}_

I added _#define DISABLE_BEACONS_ and _#define DISABLE_PING_ in my lmic_project_config.h
I also tried to verify that radio_init() called in os_init_ex() was ok but unsuccessfully

maitrekobayashi, 2021-04-19:
But I feel I didn't catch how to use properly ASSERT(0); So I added many in the code of os_init_ex and I got :

_int os_init_ex (const void *pintable) {
	ASSERT(0); **have a return of this line**
    memset(&OS, 0x00, sizeof(OS));
	ASSERT(0); **have a return of this line**
    hal_init_ex(pintable);
	ASSERT(0); **have a return of this line**
    if (! radio_init())
        return 0;;
	ASSERT(0); **DON'T have a return of this line**
    LMIC_init();
	ASSERT(0); **DON'T have a return of this line**
    return 1;;
}

void os_init() {
    if (os_init_ex((const void *)&lmic_pins))
        return;
    ASSERT(0);
}_

terrillmoore, 2021-04-19:
It appears that radio_init() is failing. 

Almost always, this is because of a wiring problem. How is your SX1272 connected to your UNO?

Best regards,
-Terry

maitrekobayashi, 2021-04-19:
Hi again , I might have found the problem, which was a pinmapping one :
The .nss pin of the transceiver was supposed to be linked to the 6th pin of the Uno, but in my case I'm using a semtech shield which physically connect the .nss pin of the SX1272 to the 10th pin of tue UNO R3.

const lmic_pinmap lmic_pins = {
//.nss = 6, //NOT WORKING
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 5,
    .dio = {2, 3, 4},
};

Thanks for your time,

Best Regards