mcci-catena/arduino-lmic #713: Make DELAY_DNW1(RX1 Delay Message) configurable in lmic_project_config

Uspizig, 2021-04-18:
I recently tried a new node on TTN EU V3 stack.
After successful OTTA I tried ABP as well.
In my code I am doing OTAA first and after a successful Join I switch to ABP. This enables a faster startup after sleep of ESP32.
After a few packets I noticed that TTN is sending a RX1 delay message to my node after every packet I transmitted.

Solution that works for me:
I changed in lorabase.h the parameter DELAY_DNW1 from 1 to 5 sec
TTN V3 requests a RX1 delay of 5 sec.
https://www.thethingsnetwork.org/forum/t/new-application-v3-otaa-not-working/43338/20

To prevent many TTN Gateways firing lots of Downlink Messages I suggest to make the parameter DELAY_DNW1 in lmic_project_config configurable as a default to 5 sec for TTN Config.

descartes, 2021-07-07:
With v2 now read-only this has become a pressing issue - whilst I think if you can implement LMIC then fishing in to lorabase.h shouldn't be too much to ask, but has tripped up many already.

Is setting a #define in to a enum acceptable? If so, I can test the change & create a PR.

piebebakke, 2021-08-25:
After the join with the network the variable  LMIC.rxDelay variable is set (to 5 for TTN V3).  

Saving this value in RTC memory and reassign this value to  LMIC.rxDelay after having woken up from deepsleep, solved the issue for me.  This seems more elegant/flexible than making it configurable.

descartes, 2021-08-31:
> After the join with the network the variable LMIC.rxDelay variable is set (to 5 for TTN V3).

But not for ABP ...

piebebakke, 2021-08-31:
This will solve the original issue of  Uspizig, who does an OTAA join, then let the ESP got deep sleep and after woken up does an ABP, based on some RTC saved LMIC fields (containing o.a. relevant data from the OTAA)

For just ABP I expect setting LMIC.rxDelay to 5 could also solve the the downlink timing issue as well.  (But I have not tested it).

descartes, 2021-08-31:
> For just ABP I expect setting LMIC.rxDelay to 5 could also solve the the downlink timing issue as well. (But I have not tested it).

Yes it will, but at present the setting is buried deep in the source files which is not ideal when trying to guide an new Arduino developer to success which is what I'm trying to get to.

I have tested it in so much as I have devices working with it for about a year now, plus appropriate configuration on TTS so that there is no need to download the other five channels & other misc settings. And configuration to ensure memory is available on smaller AVR's for one sensor devices. And sleep modes.