mcci-catena/arduino-lmic #716: Objenious : EV_JOIN_TXCOMPLETE: no JoinAccept

maitrekobayashi, 2021-04-20:
Hi again, 

I'm trying to run the ttn-otaa.ino from the latest github mainline code.
My problem seems to be a recurent one but the solutions provided didn't work. Here what i got on the terminal : 

_16:14:46.425 -> Starting
16:14:46.465 -> Packet queued
16:14:46.465 -> 1767: EV_JOINING
16:14:49.940 -> 220993: EV_TXSTART
16:14:56.960 -> 657246: EV_JOIN_TXCOMPLETE: no JoinAccept
16:15:58.439 -> 4496983: EV_TXSTART
16:16:05.432 -> 4933240: EV_JOIN_TXCOMPLETE: no JoinAccept
[...]_

Here my pinmapping :
_// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 5,
    .dio = {2, 3, 4},
};_

Here my lmic_project_config.h :
_#define CFG_eu868 1
#define DISABLE_BEACONS
#define DISABLE_PING
#define CFG_sx1272_radio 1_

Point 1 : I tried all the possibilities about lsb, msb. The DevEUI, AppEUI and AppKey I'm using here worked well with a (STM32)L073RZ+SX1272 (with Keil compiler) on this same Objenious server.
Point 2 : I tried (as you told others users to try) with adding 
_LMIC_setClockError(MAX_CLOCK_ERROR * 10 / 100);_
between the _LMIC_reset();_ and the _do_send(&sendjob);_ in the main 

So I'm currently in a lack of solutions to try, maybe someone already got the same problem ?

Thanks

Paul 

- Version of LMIC being used : last
- Version of Arduino IDE being used : 1.8.13
- Network provider : Objenious (bouygues telecom, France)
- Region : EU868
- Board : Uno R3
- Radio : Semtech SX1272 (mbed shield)

maitrekobayashi, 2021-04-21:
I MIGHT have found the problem but can't resolve it right now.
Indeed, TTN and many LoRaWAN servers use for RX2DR = 3 (SF9)
Objenious on the contrary use RX2DR = 0 (SF12).
What i don't understand is that OTAA have, for me, to adapt even if ttn-otaa.ino is designed for ttn use.
Could I modify it by myself or should I let it down and try with APB ?

terrillmoore, 2021-04-21:
I always try the raw sketch with a second radio to confirm that downlink is working at all. You might not have that option, but that's the easiest to start to get a level of confidence to build up from.

maitrekobayashi, 2021-04-22:
Hi again, 
I tried with an other semtech SX1272MB2xAS and had the same results.

terrillmoore, 2021-04-22:
Did you try two radios talking to each other with the raw sketch?

maitrekobayashi, 2021-04-22:
Yes the ping pong was working two days ago with these two SX1272 but on STM32 boards, not on Uno

miguemorales, 2021-10-07:
I'm suffering the same issues that you comment in your first message, i can send the join accept, and I also receive this join accept in my gateway, but i never receive any downlink, and I am not doing anything that may delay the timing. Have you found the problem?

terrillmoore, 2021-10-07:
Please check #798. In all cases so far, this has turned out not to be a device problem. Especially in Europe, gateways have very limited downlink bandwidth and joins may not work unless you have a gateway close by. And you must have a multi-channel gateway.

mrx23dot, 2022-02-19:
You can also try if ABP works better than OTAA.