mcci-catena/arduino-lmic #719: maxmim bytes can be send and how to see log in otaa

manivalaguru, 2021-04-26:
hello,
i want to know maximum bytes i can send in single instance whenever i tired to change this static uint8_t mydata[] = "MCCI! Node";  it gives me error i could not send more than a byte can you help me with this?

then i want see the complete log in otta mode i have inclucded this in my project_config #define LMIC_DEBUG_LEVEL 2 but that did n"t help in otaa mode i could not see anything other than this

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

but i apb mode i can see when my packets queued and also my downlink data in the console .how can i see that in ota mode too?
thanks and regards
manikandan v

**Environment**

This information is very important; it's hard to help without a complete set of answers.

- Version of LMIC being used. If using the latest github mainline, let us know, otherwise state the version.
- Version of Arduino IDE being used:1.8.12
- Network provider (The Things Network, Swisscom, ChirpStack, etc.)->chirpstack
- Region (EU868, US915, etc.)->IN865
- Board (MCCI Catena, Adafruit Feather M0, Heltec Wi-Fi LoRa 32 v2, etc.)->ESP32
- Radio (HopeRF, SX1276, etc.)->Nice rf

terrillmoore, 2021-05-31:
It's hard to answer/track multiple questions in one ticket.

1. Please avoid using debug prints. I don't support use of the library with debug prints in the LMIC because it affects timing and breaks things. The debug prints are in the code to support LMIC *development* not LMIC *use*. You're welcome to experiment with it, but bugs will not be fixed.

2. For OTAA, you must get your network server to send the channels with the CFList (on JoinAccept). TTN normally does this. Please contact chirpstack forums for help with doing this on chirpstack.  Strongly recommend you get things working on TTN first, because configuring chirpstack correctly requires substantial LoRaWAN know-how.

3. ABP is an advanced use case (even though it seems simpler) because you must change your sketch to do *all* the network settings needed to match your network server. Again, can't support that here.

Best regards,
--Terry

terrillmoore, 2021-05-31:
These are questions rather than bugs; moving to issues.

closed 2021-05-31 by terrillmoore