mcci-catena/arduino-lmic #720: Hal_sleep() is currently not implemented. Will it be?

daanvV, 2021-04-26:
It would be great if I could put the RFM95 to sleep mode since at the moment it consumes roughly 0.6 mA when not transmitting. 
In this line it shows that the hal_sleep() function exists but is not implemented. Are there any plans for it to be implemented? 
https://github.com/mcci-catena/arduino-lmic/blob/d248d762265a9302ff6fc28f722ed83f67161455/src/hal/hal.cpp#L368

Many thanks!

chansheunglong, 2021-05-25:
You use LMIC.shutdown() to turn off the lora module

terrillmoore, 2021-05-31:
Thanks for your note.

It is not likely that hal_sleep() will ever be implemented (at least, not in the form in the IBM LMIC code).

Use `os_queryTimeCriticalJobs()` and sleep the CPU for a suitable period of time once the LMIC indicates that it's idle.

Please do not call `LMIC_shutdown()`, the LMIC leaves the radio in an appropriate state.

terrillmoore, 2021-05-31:
Moving this to discussion as it's more of a question.

closed 2021-05-31 by terrillmoore