mcci-catena/arduino-lmic #738: Adding support for AS923-925 (“AS2”)

chansheunglong, 2021-05-26:
From TTN:
> AS923-925 (“AS2”) #
> Used in Brunei, Cambodia, Hong Kong, Indonesia, Laos, Taiwan, Thailand, Vietnam

The default AS923 only use 2 frequency 923.2Mhz and 923.4Mhz, but from the frequency plan info of TTN it show a total of 8 frequency (SF7 to SF12, W125) + 1 frequency (SF7, W250) + 1 FSK being use by AS923-925, will it be reasonable to add support for that by adding new band plan, e.g. "AS923_2"?

New lorabase_as923_2.c
```
enum {
  AS923_F1    = 923200000,      // g1   SF7-12
  AS923_F2    = 923400000,      // g1   SF7-12
  AS923_F3    = 923600000,      // g1   SF7-12
  AS923_F4    = 923800000,      // g1   SF7-12
  AS923_F5    = 924000000,      // g1   SF7-12
  AS923_F6    = 924200000,      // g1   SF7-12
  AS923_F7    = 924400000,      // g1   SF7-12
  AS923_F8    = 9234600000,      // g1   SF7-12
  AS923_FDOWN = 923200000,      //      (RX2 freq, DR2)
  AS923_FBCN  = 923400000,      //      default BCN, DR3
  AS923_FPING = 923400000,      //      default ping, DR3
};
```

New lmic_as923_2.c
```
enum { NUM_DEFAULT_CHANNELS = 8 };
static CONST_TABLE(u4_t, iniChannelFreq)[NUM_DEFAULT_CHANNELS] = {
        // Default operational frequencies
        AS923_F1 | BAND_CENTI,
        AS923_F2 | BAND_CENTI,
        AS923_F3 | BAND_CENTI,
        AS923_F4 | BAND_CENTI,
        AS923_F5 | BAND_CENTI,
        AS923_F6 | BAND_CENTI,
        AS923_F7 | BAND_CENTI,
        AS923_F8 | BAND_CENTI,
};
```

Rest of the file should be basically the same as the current AS923 config...

Thanks!

chansheunglong, 2021-05-26:
After reading the specification, I believe using "**LMIC_setupChannel**" in the main code to set up additional channels is a better choice? This will keep LMIC stick to the LoRaWan spec while allowing additional channels.

closed 2021-05-26 by chansheunglong

IoTThinks, 2021-05-26:
Correction for LoRaWAN Frequency in Vietnam. 
We will not use 923-925Mhz by law and LoRaWAN puts us in 920-923Mhz (less than 923Mhz)
* LoRaWAN renamed the orginainal AS923 as AS923-1.
* And AS923-2 is for countries like Vietnam with frequencies less than 923Mhz.

This is the proposal for AS923-2.
https://github.com/TheThingsNetwork/lorawan-frequency-plans/pull/30

The updated frequency plan by TTN for Asia is here
https://www.thethingsindustries.com/docs/reference/frequency-plans/#as_920_923

chansheunglong, 2021-05-26:
> Correction for LoRaWAN Frequency in Vietnam.
> We are not allowed to use 923-925Mhz in Vietnam by law and LoRaWAN puts Vietnam in 920-923Mhz (less than 923Mhz)
> 
> * LoRaWAN renamed the orginal AS923 as AS923-1.
> * And AS923-2 is for countries like Vietnam with frequencies less than 923Mhz.
> 
> The updated frequency plan by TTN for Asia is here
> https://www.thethingsindustries.com/docs/reference/frequency-plans/#as_920_923
> 
> This is the proposal for AS923-2 for Vietnam.
> [TheThingsNetwork/lorawan-frequency-plans#30](https://github.com/TheThingsNetwork/lorawan-frequency-plans/pull/30)

Thanks for the updated information!