mcci-catena/arduino-lmic #769: LoRaWAN 1.0.4 support

maxgerhardt, 2021-07-09:
**Is your feature request related to a problem? Please describe.**

Currently, LMIC supports, [per README](https://github.com/mcci-catena/arduino-lmic#selecting-the-lorawan-version), LoRaWAN versions 1.0.2 and 1.0.3. Per https://www.thethingsnetwork.org/article/whats-new-in-lorawan-104-1 and https://lora-alliance.org/resource_hub/lorawan-104-specification-package/, LoRaWAN version 1.0.4 has been released in 2020, with multiple security improvements:
* enforcing up/downlink counters to be 32-bit wide and saved persistently in non-volatile memory
* changes to how the DevNonce is chosen for OTAA (monotonic increase instead of random choice)
* AppEUI & AppNonce replaced with JoinEUI & JoinNonce
* etc.

T
**Describe the solution you'd like**
It would be nice for (Arduino-) LMIC to officially support LoRaWAN v1.0.4.

**Describe alternatives you've considered**
Stay at 1.0.3 with these security improvements not taken advantage of. 

**Additional context**
None

descartes, 2021-07-10:
Yeah, it would, as a nice to have if your devices are doing top secret work but it's not vital to security and it's definitely non-trivial - how will you be assisting with this?

mrx23dot, 2022-02-28:
I would definitely use a session struct inside LMIC, containing everything needed for activation and ABP, including counters.
Then it can be easily saved/restored to/from NV, even swapping network operators.

Using random nonce is better if NV fails.
Websites still use old naming conventions first those have to change.

HeadBoffin, 2022-02-28:
> Using random nonce is better if NV fails. 

Excepting that v1.0.4 requires an incrementing JoinNonce ...

> Websites still use old naming conventions first those have to change.

Why? v1.0.2 & v1.0.3 devices are very prevalent so the nomenclature is correct for them.

terrillmoore, 2026-10-19:
To do for 1.0.4, DevNonce: TS001-1.0.4 section 6.2.5 makes DevNonce a counter that starts at 0 at first power-up, goes up by one per Join-Request, never repeats for a JoinEUI, and is kept across power cycles; the device stops joining when all 65536 values are used (TR001 Remedy #2: change JoinEUI). See #1156. The LMIC is 1.0.3, so #1157 keeps a random start, counts up from it and wraps, which a 1.0.3 server accepts because it rejects only nonces it remembers. The 1.0.4 counter (start at 0, persist in the session-state blob, stop at exhaustion) is left for 1.0.4 support.

terrillmoore, 2026-10-20:
Progress toward 1.0.4 in #1157 (v7-devel): DevNonce is now a start value plus a count of Join-Requests, kept across LMIC_reset() and in session state V3, with a stop when every value is used (EV_DEVNONCE_EXHAUSTED). For 1.0.4 the same counter would start at 0; that switch isn't made yet, since the LMIC is still 1.0.3. See also #1156.