mcci-catena/arduino-lmic #771: Using Arduino LMIC library with EnableInterrupt

j45p41, 2021-07-21:
I am using the EnableInterrupt library, infact the only library that I can find that allows me to use six interrupts to wake the Arduino. However when I use it with the LMIC library (Matthijs Kooijman ) the "Forward uplink data message" is not sent during initial registration, only if #include <EnableInterrupt.h> is commented out does it work. I am using a RFM95 with the following LMIC pin assignments:

const lmic_pinmap lmic_pins = { .nss = 10, .rxtx = LMIC_UNUSED_PIN, .rst = 9, .dio = {3, 8, LMIC_UNUSED_PIN}, };

This must be something to do with the library sharing ISRs. Could somebody point to which parts of the LMIC library so as not to have this conflict?

I have tried swapping 8 for A5 and it again only works when the line is commented. Any help would be greatly appreciated.

Please give a clear and concise description of the problem you're facing and what you'd like help with.

- Version of LMIC - LATEST
- Version of Arduino IDE being used: platformio Core 5.1.1·Home 3.3.4
- Network provider The Things Network
- Region EU868
- Board (Atmega328p)
- Radio (HopeRFM95)

terrillmoore, 2021-07-21:
Sorry you're having trouble.

I inherited the "interrupt driven" option, but I have never tested it or used it. Plans are to extensively rework the LMIC around timing and HAL interfaces, at which point all the interrupt driven stuff is likely to change. So although we have the feature, it's definitely for advanced users who are able to debug any problems encountered. My recommendation: test the LMIC without this using this feature, get your code working, then enable this and debug it as a last step.

j45p41, 2021-07-21:
Thank you for the suggestion:
I have managed to get it to work by specifying
#define EI_NOTPORTD however it does not work with  #define EI_ARDUINO_INTERRUPTED_PIN which is needed for my project to function. I will keep investigatibg..

j45p41, 2021-07-22:
This is the best I can get to work with 6 available pin change interrupts - just enough! 

#define EI_ARDUINO_INTERRUPTED_PIN
#define EI_NOTPORTB  // DISABLE FOR LMIC
#define EI_NOTPORTD  // DISABLE FOR LMIC