mcci-catena/arduino-lmic #776: Customize the join procedure

Kampi, 2021-08-01:
**Describe your question or issue**

I use the current version of the MCCI LoRaWAN library with an ESP32 MCU. As far as I observed it correctly the library is waiting around 1 minute between each join request by default and the device isn´t going to any kind of power-saving mode.

I´m looking for a way to reduce the wait time between two join requests. And I´m looking also for a way to put the MCU into sleep during each join request. Is there a chance to do this?

**Environment**

- ESP32
- Latest version from GitHub
- PlattformIO
- TTN
- EU868
- Custom
- HopeRF SX1276

chansheunglong, 2021-08-03:
You can always join by calling `LMIC_startJoining()`, but you have to make sure you are following the duty cycle (check your local law) and fair use policy (e.g. from TTN). For putting MCU to sleep, you should ask help in Espressif's forum.

But since I also work with ESP32 board...
If you go for deep sleep mode, ESP32 will be reset when waking up (all ram content lost). If the first thing your program does is to OTAA, you can just instruct the MCU to sleep for 1 minute then wake up, and start all over again.
```
esp_sleep_enable_timer_wakeup(60000000); //Sleep for 60 seconds
esp_deep_sleep_start();
```

miguemorales, 2021-10-06:
Hi @Kampi, I am using almost the same set up as you, and sometimes my end-nodes are pretty far from the gateway, having joining issues and waiting some iterations to get the downlink even when in the gateway I oftenly receive the joint-resquest with no problems but por rssi and SNR.
Did you get to request the joint in a higher frequency? Can you explain me what was your solution.
Thank you in advance

closed 2022-04-06 by Kampi