mcci-catena/arduino-lmic #779: Arduino Pro Mini, RFM95 Compilation error, Gloval Variable Exceeded 2048 Bytes

dagnolof, 2021-08-17:
Arduino Pro Mini Atmega328P 3.3V 8MZ
Arduino IDE Version 1.8.15
MCCI LoRaWAN LMIC Library Version 4.0.0

I am trying to setup a lorawan device using Arduino Pro Mini and RFM95 module according to 
tutorial https://www.mobilefish.com/developer/lorawan/lorawan_quickguide_build_lora_node_rfm95_arduino_pro_mini_v3.html

However i can not compile the code.

I get following error
`data section exceeds available space in boardSketch uses 25688 bytes (83%) of program storage space. Maximum is 30720 bytes.

Global variables use 2331 bytes (113%) of dynamic memory, leaving -283 bytes for local variables. Maximum is 2048 bytes.
Not enough memory; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing your footprint.
Error compiling for board Arduino Pro or Pro Mini.
`

I am struggling with this issue for a while now.

Can you please give me an indication on how to proceed with this problem

This is the code i am using.
It is base on ttn-otaa example

`/*******************************************************************************
 * Copyright (c) 2015 Thomas Telkamp and Matthijs Kooijman
 * Copyright (c) 2018 Terry Moore, MCCI
 *
 * Permission is hereby granted, free of charge, to anyone
 * obtaining a copy of this document and accompanying files,
 * to do whatever they want with them without any restriction,
 * including, but not limited to, copying, modification and redistribution.
 * NO WARRANTY OF ANY KIND IS PROVIDED.
 *
 * This example sends a valid LoRaWAN packet with payload "Hello,
 * world!", using frequency and encryption settings matching those of
 * the The Things Network.
 *
 * This uses OTAA (Over-the-air activation), where where a DevEUI and
 * application key is configured, which are used in an over-the-air
 * activation procedure where a DevAddr and session keys are
 * assigned/generated for use with all further communication.
 *
 * Note: LoRaWAN per sub-band duty-cycle limitation is enforced (1% in
 * g1, 0.1% in g2), but not the TTN fair usage policy (which is probably
 * violated by this sketch when left running for longer)!

 * To use this sketch, first register your application and device with
 * the things network, to set or generate an AppEUI, DevEUI and AppKey.
 * Multiple devices can use the same AppEUI, but each device has its own
 * DevEUI and AppKey.
 *
 * Do not forget to define the radio type correctly in
 * arduino-lmic/project_config/lmic_project_config.h or from your BOARDS.txt.
 *
 *******************************************************************************/

#include <arduino_lmic.h>
#include <hal/hal.h>
#include <SPI.h>

//
// For normal use, we require that you edit the sketch to replace FILLMEIN
// with values assigned by the TTN console. However, for regression tests,
// we want to be able to compile these scripts. The regression tests define
// COMPILE_REGRESSION_TEST, and in that case we define FILLMEIN to a non-
// working but innocuous value.
//
#ifdef COMPILE_REGRESSION_TEST
# define FILLMEIN 0
[... 182 lines cut ...]
    // For Pinoccio Scout boards
    pinMode(VCC_ENABLE, OUTPUT);
    digitalWrite(VCC_ENABLE, HIGH);
    delay(1000);
    #endif

    // LMIC init
    os_init();
    // Reset the MAC state. Session and pending data transfers will be discarded.
    LMIC_reset();

    // Use with Arduino Pro Mini ATmega328P 3.3V 8 MHz
    // Let LMIC compensate for +/- 1% clock error
    LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);
 
    // Start job (sending automatically starts OTAA too)
    do_send(&sendjob); 
}

void loop() {
    os_runloop_once();
}`

Thanks in advance,
Fabrizio

cstratton, 2021-09-23:
Getting a modern LMiC to fit in an ATmega328p is a bit challenging, it's not really a recommended target platform.

However, you have an additional issue in that your particular board has a larger than needed bootloader, leaving less space for the sketch code.  Compare these lines of Arduino boards.txt:

pro.menu.cpu.8MHzatmega328.upload.maximum_size=30720

vs

uno.upload.maximum_size=32256

Which reveal that more optimal bootloaders for the ATmega328p like that used on the Uno would leave more space, possibly just barely enough to fit modern LMiC.

If you have an ISP setup (possibly another Arduino running the ISP sketch) it might be possible to change the bootloader and either start pretending your pro mini is some other board, or even create a custom boards.txt entry giving the altered bootloader and size.  But that's not going to be simple.

If you want to work with LoRaWan on an ongoing basis you should really change host platforms to one of those better supported.

terrillmoore, 2021-09-23:
Thanks, Chris (good to see you)!

@dagnolof, you should look at the configuration I use for CI testing. I think you need to change a few config settings to reduce the RAM usage. The most important one is to reduce the maximum message size to 64 bytes (or smaller); the LMIC has fixed buffers that are 2x that number of bytes, and the default is 240.  So it saves a lot of RAM. Also turn off ClassB support.