mcci-catena/arduino-lmic #783: LoPy4 support

triple-it, 2021-08-31:
**Describe your question or issue**
Hi,
I would like to know if the Pycom LoPy4 is easy supported with this library, or that we should develop some custom Hal Configuration for this board first.

[image]

This the HLD overview

[image]

**Environment**

This information is very important; it's hard to help without a complete set of answers.

- Version of LMIC being used. If using the latest github mainline, let us know, otherwise state the version.
  MCCI LoRaWAN LMIC Library 4.0.0
- Version of Arduino IDE being used:
  1.8.13
- Network provider (The Things Network, Swisscom, ChirpStack, etc.)
  TTN
- Region (EU868, US915, etc.)
  EU868
- Board (MCCI Catena, Adafruit Feather M0, Heltec Wi-Fi LoRa 32 v2, etc.)
   PyCom LoPy4: https://docs.pycom.io/gitbook/assets/specsheets/Pycom_002_Specsheets_LoPy4_v2.pdf
- Radio (HopeRF, SX1276, etc.)
  SX1276

The old LoPy (not 4) with IBM LMIC Framework has been reported to work with the following:
```
const lmic_pinmap lmic_pins = {
    .mosi = 27,
    .miso = 19,
    .sck = 5,
    .nss = 17,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 18,
    .dio = {23, 23, 23}, //workaround to use 1 pin for all 3 radio dio pins
};
```

For the LoPy4 I am using

```
#define SS 18 // ESP32 GPIO18 (Pin18) – SX1276 NSS (Pin19) SPI Chip Select Input
#define MOSI 27 // ESP32 GPIO27 (Pin27) – SX1276 MOSI (Pin18) SPI Data Input
#define MISO 19 // ESP32 GPIO19 (Pin19) – SX1276 MISO (Pin17) SPI Data Output
#define SCK 14 // ESP32 GPIO5 (Pin5) – SX1276 SCK (Pin16) SPI Clock Input
#define RST 27 // ESP32 GPIO14 (Pin14) – SX1276 NRESET (Pin7) Reset Trigger Input
#define DIO0 23 // ESP32 GPIO26 (Pin15) – SX1276 DIO0 (Pin8) used by LMIC for IRQ RX_Done & TX_Done
#define DIO1 23 // ESP32 GPIO33 (Pin13) – SX1276 DIO1 (Pin9) used by LMIC for IRQ RX_Timeout
#define DIO2 23 // ESP32 GPIO32 (Pin12) – SX1276 DIO2 (Pin10) not used by LMIC
#define Vext 21

const lmic_pinmap lmic_pins = {
  .nss = SS,
  .rxtx = LMIC_UNUSED_PIN,
  .rst = RST,
  .dio = {DIO0, DIO1, DIO2},
};
```

But this is the result on a LoPy4

st:0x1 (POWERON_RESET),boot:0x32 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
Starting
nss: 18
rst: 27
dio[0]: 23
dio[1]: 23
dio[2]: 23
FAILURE 
C:\Users\r.groesbeek\OneDrive - Triple IT BV\Documents\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src\lmic\oslmic.c:53

So, what would be a working configuration for the LoPy4?

lnlp, 2021-09-16:
> I would like to know if the Pycom LoPy4 is easy supported with this library, or that we should develop some custom Hal Configuration for this board first.

MCCI's LMIC library will run on the Pycom LoPy4.

For a working example see **LMIC-node** (https://github.com/lnlp/LMIC-node).
LMIC-node includes support for the LoPy4 and helps you get your node quickly up and running.

Do read LMIC-node's README.md and the included BSF file for the LoPy4 board.
The board is not very C/C++ development friendly and uploading of the firmware needs some specific steps which are documented in above mentioned documents. 

In the BSF file you can find additional useful information about pin mappings etc.

lnlp, 2021-12-05:
@terrillmoore 
This issue can be closed. It was already answered above with a link to a working example: LMIC-node.
No response from @triple-it which is sad and makes this look wasted time.