mcci-catena/arduino-lmic #784: Add check to loop/os_runloop_once to ensure valid state

olicooper, 2021-09-09:
**Is your feature request related to a problem? Please describe.**
I wanted to disable LMIC so I commented out the `Arduino_LoRaWAN::begin()` function in Arduino's `setup()` function. I compiled and ran the code - which resulted in a backtrace on the serial output which I had to decode to understand what went wrong.

**Describe the solution you'd like**
Could there be more graceful handling of LMIC being in an uninitialized state? For example an `if` statement that skips the loop functionality until `begin()` has been called?

olicooper, 2021-09-17:
An update...
The exception source resides in the call to `hal_processPendingIRQs` inside `os_runloop_once`
I have the fix here: https://github.com/olicooper/arduino-lmic/commit/172a1a8990d80813c208861ba09781bffe58bd3e

This stops the kernel panic from happening if `os_runloop_once` is called without `os_init` being called beforehand.

The reason I need this fixing is because I conditionally switch between lmic and mqtt depending on service availability, but because `os_setTimedCallback` is required to ensure correct timings when lmic is running, I also need it to function while lmic isn't running too. A better approach would be to abstract the timings to an external library (e.g. use FreeRTOS round-robin implementation) but that's likely too much work and I don't know how I'd do that right now.

terrillmoore, 2021-10-17:
My recommendation is *always* to init the LMIC, but stop it when you're not using it. Call `LMIC_shutdown()` immediately after `LMIC_init()` during startup, and whenever you want to stop it thereafter. Then call `LMIC_reset()` to restart it.

closed 2021-10-17 by olicooper