mcci-catena/arduino-lmic #785: Gateway receving uplink message, but end device shows last info unavaible

JoaoPedroCAS, 2021-09-09:
I'm pretty noob at this, but i'm using 2 heltec esp32 lora, one as a single channel gateway (https://github.com/things4u/ESP-1ch-Gateway) and other one as the end device (my college gave me this 2 devices to make it work). 

The gateway receive the uplink message, the device addres is equal to what is defined in the end device page, but the end device page never changes the status. What could it be?

Thank you in advance

cstratton, 2021-09-23:
What are you using as a network server?

Single/dual channel gateways are obsolete and heavily discouraged - TTN for example **outright bans them.**

G-3r4-d, 2021-09-30:
> I'm pretty noob at this, but i'm using 2 heltec esp32 lora, one as a single channel gateway (https://github.com/things4u/ESP-1ch-Gateway) and other one as the end device (my college gave me this 2 [...]

I'm in the same situation. I tried otaa and abp example codes. The otaa example file has connetion with TTN v3, I'm not going to recieve any downlink messages because I'm also using a single channel gateway with the same library, I just needed to try if my gateway works. 

But ABP example file don't connect with TTN, I entered correctly the NwkSkey and AppSkey in msb as the example says, but TTN does not take my messages. The  mType field says "UNCONFIRMED UPLINK".

The gateway recieve a message with "UnConfirmed uplink"

G-3r4-d, 2021-09-30:
> Again, **TTN absolutely prohibits** the connection of "single channel gateways" (and dual channel ones) to their network, so please disconnect that. If you really want to use one, you're going to need to use a private network server. TTN may only be used with gateways based on proper multi-channel/mutli-SF concentrator cards.

I appreciate your comment, do you have any documentation about how to create my own private LoRaWAN network server? I'm new at this field and do not have any experience in this.

cstratton, 2021-09-30:
People typically use Chirpstack (or in the case of a few gateway manufacturers who package it onboard, a very old version from before the forced name change from loraserver)

G-3r4-d, 2021-10-01:
Thanks pal. I will give it a try