mcci-catena/arduino-lmic #787: Possibly suboptimal LORA single receive mode?

zhgzhg, 2021-09-15:
Greetings,
The datasheets of the SX127x series and the corresponding application notes suggest modifying RegInvertIQ2 when receiving LORA downlinks. I couldn't find any traces of that insinde _src/lmic/radio.c_, thus I wanted to address it.

The docs:

[SX1276-7-8-9 Datasheet](https://semtech.my.salesforce.com/sfc/p/E0000000JelG/a/2R0000001Rbr/6EfVZUorrpoKFfvaF_Fkpgp5kzjiNyiAbqcpqh9qSjE), the end of page 115
[SX1272 Datasheet](https://semtech.my.salesforce.com/sfc/p/E0000000JelG/a/440000001NCE/v_VBhk1IolDgxwwnOpcS_vTFxPfSEPQbuneK3mWsXlU), the end of page 113
[AN 1200.24](https://semtech.my.salesforce.com/sfc/p/E0000000JelG/a/440000001NAw/7YN8ZamV70_xR.vPDAAAshm.0Wt4jmRX0nOKkOzQqiI), the end of page 6

**Environment**

- Version of LMIC being used. v4.0.0

cstratton, 2021-09-23:
Not sure what you think needs to be done "to address it" - the code which configures the needed IQ inversion mode is right here:

https://github.com/mcci-catena/arduino-lmic/blob/master/src/lmic/radio.c#L941

zhgzhg, 2021-09-23:
@cstratton ,
at https://github.com/mcci-catena/arduino-lmic/blob/master/src/lmic/radio.c#L941 I can see only RegInvertIQ as being modified.
The documentation from the links I've given above suggest modifying RegInvertIQ**2** as well. I'm not sure if the implementation deliberately is not working with RegInvertIQ**2**, or it has been missed.

cstratton, 2021-09-24:
Ah, ok, sorry missed the IRQ2 part.  It's kind of odd that they've only revealed the details of these chips bit by bit.

I see that the LoRaMac-node code does set this, probably worthwhile to adopt that

https://github.com/Lora-net/LoRaMac-node/blob/master/src/radio/sx1276/sx1276.c#L842

Have you tried adding this to LMiC?

terrillmoore, 2021-09-24:
@zhgzhg the datasheet I have does not mention this register or these bits. Thanks for bringing this to our attention, but @cstratton is right; this code was all implemented long before that datasheet change was announced.

zhgzhg, 2021-09-24:
Got it, thanks @terrillmoore .

Yes, I've tried modifying the register in my local LMiC copy, @cstratton . I couldn't notice any negative effects, but at the moment I don't have proper test environment to say confidently.

terrillmoore, 2022-02-28:
Curiously, that's in the *transmit* path. Our use of the noIqInversion is in the RX path.  The sample code doesn't seem to use noIqInversion during receive! (And the default is, I think, in the proper state.)