mcci-catena/arduino-lmic #792: TTN v3 don't recieved ABP uplinks messages (UNCONFIRMED UPLINK)

G-3r4-d, 2021-09-30:
**Consider raising support questions on the forum first**

There is a discussion site, [forum.mcci.io](https://forum.mcci.io), which includes a [category focused on the Arduino LMIC](https://forum.mcci.io/c/device-software/arduino-lmic/5). Unless you're sure that your problem is a bug in the LMIC, it would be great if you can try for help there first. That will keep issues focused on work that needs to be done by the developers.

**Describe your question or issue**

Please give a clear and concise description of the problem you're facing and what you'd like help with.

**Environment**

This information is very important; it's hard to help without a complete set of answers.

- Version of LMIC being used. If using the latest github mainline, let us know, otherwise state the version.
- Version of Arduino IDE being used:
- Network provider (The Things Network, Swisscom, ChirpStack, etc.)
- Region (EU868, US915, etc.)
- Board (MCCI Catena, Adafruit Feather M0, Heltec Wi-Fi LoRa 32 v2, etc.)
- Radio (HopeRF, SX1276, etc.)

cstratton, 2021-10-01:
>TTN V3 using another ESP32 LoRa V2 as a single channel gateway using this library https://github.com/things4u/ESP-1ch-Gateway 

As explained at https://github.com/mcci-catena/arduino-lmic/issues/785 since channel "gateways" are prohibited by TTN

cyberman54, 2021-12-28:
This can (and should be) closed.

terrillmoore, 2021-12-28:
Moving to a discussion.