mcci-catena/arduino-lmic #794: Joining TTN over OTAA fails (EV_JOIN_TXCOMPLETE: no JoinAccept)

T94T, 2021-10-01:
I'm using the latest arduino-lmic library on an Arduino compatible chip running at 16MHz, LMIC_LORAWAN_SPEC_VERSION_1_0_3, CFG_eu868 and CFG_sx1276_radio

When trying to send a payload with the _TTN-OTAA_ script it quits with the board's output message: `EV_JOIN_TXCOMPLETE: no JoinAccept`

But in the TTN console I get: `Accept join-request`
[image]

To be sure the downlink on the board also works, I ran the _RAW_ example on two dev boards. The test message can be sent and received in both ways.

To further narrow down the problem, I tried the whole thing in different geographic locations.

terrillmoore, 2021-10-01:
Thank you for running the tests.

I suspect a clock accuracy problem. The candidate problems are:

1. RF receive chain on device -- ruled out by your experiments.
2. Gateway issues with not having enough bandwidth to tx the join-accept due to compliance issues. Possibly ruled out by your experiments with multiple gateways; but not necessarily, depending on the traffic to other devices and downlink packet loads.
3. Clock accuracy on the device -- not ruled out at all, and a known issue. You don't mention your device or how it is clocked.
4. LMIC bug -- unlikely but not impossible
5. Bug in timing in your application -- since you're using the TTN OTAA sketch, this seems unlikely.
6. Device antenna problems -- not ruled out.

To rule out 3, please edit your project config header file.

Add: 
```c++
#define LMIC_ENABLE_arbitrary_clock_error 1
```

Then, in setup(), after initializing the LMIC, call:

```c++
LMIC_setClockError(MAX_CLOCK_ERROR * 5 / 100);
```

This sets the clock error to 5%, causing the LMIC to open the RX window for downlinks much earlier, and to leave it open much longer. (Without `LMIC_ENABLE_arbitrary_clock_error`, the LMIC limits the error to 0.4%, no matter what you set.)  Usually 5% is enough to deal with gross issues -- but 5% of 5 seconds is only 250 ms, so you might have to try 10%.  

If nothing helps here, then you have to go back and look at the RSSI for downlinks. It's very common for gateways to be able to hear devices, but devices can't hear gateways.  

Devices will hear each other even without working antennas, if they're side by side. The same is true for devices and gateways.

But if the device antenna is not functional, once the device is not in the same room with the gateway, you'll have problems.

If you can find the RSSI and SNR numbers in the "received join request" log on the console -- you may have to turn on verbose logging -- you can get a clue. I had a device which was not getting join accepts; the join requests were at about -120 dB and negative SNR.  I had to put a gateway closer to the device.

Clearly, better diagnostics would help, but... time and budget don't allow me to look at that right now.

terrillmoore, 2021-10-01:
No evidence of an LMIC bug, so removing the bug label.

T94T, 2021-10-02:
Thank you very much for your detailed feedback!
We tested the `#define LMIC_ENABLE_arbitrary_clock_error 1` and `LMIC_setClockError(MAX_CLOCK_ERROR * 5 / 100);` but still the same issue. Tested also with another antenna. No luck.
To reach a solution for this as soon as possible we just ordered an 8 channel gateway... Will keep you updated next week.

T94T, 2021-10-02:
By the way: When using the exact same setup in ABP mode the `LMIC.seqnoDn` gets updated. Does this mean that downlink works?
We found a value ratio of up and down:
```
LMIC.seqnoUp = 3885
LMIC.seqnoDn = 3000
```

closed 2021-10-07 by T94T

T94T, 2021-10-07:
Just tried it. `#define LMIC_ENABLE_arbitrary_clock_error 1` does not work.