mcci-catena/arduino-lmic #806: FAILURE

CHMSFlix, 2021-10-14:
Hi, 

I getting this error while I run the code in Arduino UNO. using ttn_abp.ino

Serial Monitor. 
Error:  Starting
 7143: EV_TXSTART
Packet queued
FAILURE 
C:\Users\issel\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:923

lmic_pins configuration is: 

const lmic_pinmap lmic_pins = {
    .nss = 10,                       // chip select on feather (rf95module) CS
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 9,                       // reset pin
    .dio = {2, 6, 7}, 
};
I checked the connection. It's perfectly connected. let me know what will be the problem.

Please help to resolve this.

Environment:-

- Version of Arduino IDE being used: 1.8.15
- Network provider (The Things Network, Swisscom, ChirpStack, etc.) TTN
- Region (EU868, US915, etc.) EU868
- Board (MCCI Catena, Adafruit Feather M0, Heltec Wi-Fi LoRa 32 v2, etc.) 
- Radio (HopeRF, SX1276, etc.) HpDTek V1.1.
-

terrillmoore, 2021-10-14:
Sorry you're having problems. The failure points us to line 926 of radio.c. There we see:

https://github.com/mcci-catena/arduino-lmic/blob/26326e4fc30fd6ee95cf8d5f78e16f34e957d38b/src/lmic/radio.c#L922-L923

This means.
1. The LMIC sent a SPI command to put the radio into LoRa mode.
2.  The LMIC read back the mode register, and the radio is not in LoRa mode.

This is not anything software can recover from, as it indicates unspecified behavior in the SX1276. It will be necessary to investigate the hardware connections more carefully.

I do not know your background, and you haven't posted photos of your setup. My recommendations therefore may seem either too simple or too complicated. Apologies in advance.

My guess is grounding or power supply issues. Check your grounds, make sure you have adequate decoupling, make sure you have short wires. Use an oscilliscope to look at the AC noise on Vdd and GND while operating the radio. Modify the code at line 923 so that it loops for ever doing `opmodeLoRa()` followed by `readReg(RegOpMode)` -- nothign else in a `while(1)` loop. Then you can use a scope to look at signal quality on the SPI clk and data lines as well as Vdd and gnd.

Good luck!

CHMSFlix, 2021-10-18:
HiTerrillmorre, 
Thanks for the reply, 

Apologies for the late, I have gone through the hardware connection it is perfectly connected. now I am getting the problem here. 

 Startin801
Starting
7143: EV_TXSTART
Packet queued
FAILURE 
 C:\Users\issel\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:1065. 
i.e  
static void startrx (u1_t rxmode) {
    ASSERT( (readReg(RegOpMode) & OPMODE_MASK) == OPMODE_SLEEP );
    if(getSf(LMIC.rps) == FSK) { // FSK modem
        rxfsk(rxmode);
    } else { // LoRa modem
        rxlora(rxmode);
    }
}

mrx23dot, 2022-02-16:
I wouldn't put assertion onto external sources, 
noise on SPI can bring down the whole system (e.g. pump controller).

All it should do is to start a comms reinit.

Field experience: once in a while a MEMS device got into an undocumented state, which should have never happened, but after a reinit it works for a long time again.