mcci-catena/arduino-lmic #813: Not receiving any packets at gateway in CN490 region

wong-hao, 2021-10-30:
**Describe your question or issue**
I follow the tutorial and create files to support CN490 region. After running the scratch successfully, nothing can be received.
Any way to print the details of the sent packet such as frequency, content, and other information?

**Environment**

- Version of LMIC being used: latest
- Version of Arduino IDE being used: latest
- Network provider (Chirpstack)
- Region (CN490)
- Board (Dragino shield)
- Radio (HopeRF)

terrillmoore, 2021-10-31:
For the space overflow issue, please set DISABLE_PING and DISABLE_BEACON to 1 in your configuration.

Also, recommend checking the RAW sketch first with two devices, if possible -- that way you can confirm that the radio driver is working in the new band.

wong-hao, 2021-11-01:
> For the space overflow issue, please set DISABLE_PING and DISABLE_BEACON to 1 in your configuration.
> 
> Also, recommend checking the RAW sketch first with two devices, if possible -- that way you can confirm that the radio driver is working in the new band.

After I disable the class B and set debug to 1, the serial monitor in Arduino IDE infinitely repeats the report "starting", is there something wrong?

terrillmoore, 2021-11-01:
> After I disable the class B and set debug to 1, the serial monitor in Arduino IDE infinitely repeats the report "starting", is there something wrong?

Which sketch are you using?

wong-hao, 2021-11-02:
> > After I disable the class B and set debug to 1, the serial monitor in Arduino IDE infinitely repeats the report "starting", is there something wrong?
> 
> Which sketch are you using?

Exactly this [one](https://github.com/mcci-catena/arduino-lmic/blob/master/examples/ttn-abp/ttn-abp.ino). By the way, even I disable everything, the IDE still warns that the global variable almost uses the whole dynamic memory when debugging, I guess this is the reason.
When I cancel debugging, no such phenomenon.

terrillmoore, 2021-11-02:
Possibly running out of stack space. Try setting the maximum message size smaller: `-D LMIC_ENABLE_long_messages=0` or the equivalent. This will let you debug, probably.

closed 2021-11-03 by wong-hao