mcci-catena/arduino-lmic #814: Dear sir, how to get the txpower level at the transmitted packet?

wong-hao, 2021-11-06:
**Describe your question or issue**

When set **LMIC_DEBUG_LEVEL** we can get the frequency information, but how to get the power?
Because I am testing the mac command about ADR, I need to know that the power has changed and has changed to the correct value.

terrillmoore, 2021-11-06:
Capture it at EV_TXSTART time. Look at how the compliance sketch does this.

terrillmoore, 2021-11-06:
By the way: please _do not use_ `LMIC_DEBUG_LEVEL` except to debug crashes.  It ruins timing, and therefore the LMIC will not behave correctly.  I have not wanted to remove it, but it's not useful except for crash tracing. It is **especially** useless for looking at ADR performance and anything else that requires downlinks to work.

I am not exaggerating.

You must never print while the LMIC is doing critical work, and if you're in the LMIC, you're doing critical work. You must *log*, and then print after the LMIC finishes up. The compliance sketch is currently the best example.

Thanks,
--Terry

wong-hao, 2021-11-07:
> By the way: please _do not use_ except to debug crashes. It ruins timing, and therefore the LMIC will not behave correctly. I have not wanted to remove it, but it's not useful except for crash tracing. It is **especially** useless for looking at ADR performance and anything else that requires downlinks to work.`LMIC_DEBUG_LEVEL`
> 
> I am not exaggerating.
> 
> You must never print while the LMIC is doing critical work, and if you're in the LMIC, you're doing critical work. You must _log_, and then print after the LMIC finishes up. The compliance sketch is currently the best example.
> 
> Thanks, --Terry

Thx sir! You are So HELPFUL! :)

closed 2021-11-07 by wong-hao