mcci-catena/arduino-lmic #815: LMIC.freq does not contain frequency anymore (after updating to 4.1.0)

hb9eue, 2021-11-06:
Hi

Today I updated some of my code to use LMIC_setAdrMode(1) and commented out: LMIC_setLinkCheckMode(0) as, reading the docs, this is not used anymore, and as I understand is now the recommended opmode for TTNv3.

After each TX (event EV:TXCOMPLETE) I record Frequency and SF to log/display:

```
sprintf(s,"SF %d ", getSf(LMIC.rps) + 6);
sprintf(s,"F %.1f ", ((float)LMIC.freq / (float)1000000));
```

But LMIC.freq is 0.

Any hint what might have gone wrong? I also update all libraries/board definitions involved so maybe it's not even the fault of the LMIC library.

-Benoit-

hb9eue, 2021-11-06:
Did some more investigations. The whole LMIC data structure seems to be missing. Noticed as I am storing this to RTC to restore it after deep-sleep on the ESP32.

hb9eue, 2021-11-06:
Ok, found the cause. After reading the docs, I disabled ping and beacons as I assumed I would not need them on my Class A device.
Turn out, this is what caused the LMIC data I was interested in, to be partially missing. I don't understand why, when I read struct lmic_t I don't see for example the frequency being removed from the structure by disabling BEACONS or PING. But this is what caused my issue.
I wonder if someone could explain.

terrillmoore, 2021-11-06:
`LMIC.freq` is used everywhere and not conditionally compiled. So it's not gone, but your code isn't finding it. Perhaps you have a build problem (maybe even due to needing a clean build?)