mcci-catena/arduino-lmic #818: ttn-otaa-network-time.ino returns with USER CALLBACK: Not a success

geronimoooooooo, 2021-11-19:
I need to get time to my device ttgo t-beam and therefore I used ttn-otaa-network-time.ino. 
https://github.com/mcci-catena/arduino-lmic/blob/master/examples/ttn-otaa-network-time/ttn-otaa-network-time.ino

But I never pass  

   if (flagSuccess != 1) {      
        Serial.println(F("USER CALLBACK: Not a success"));
        return;
    } 

flagSuccess has always value 0.
The communication with TTN v3 works fine and there are no other problems.

latest mcci arduino-lmic
platformIO
Regio EU868
ttgo t-beam v1.1

Does this example work with TTN v3? Do I have to set up anything special to be able to query time?

any help would be appreciated

dhineshkumarmcci, 2021-11-19:
yes, this example work with TTN V3.

> Do I have to set up anything special to be able to query time?

Yes, you have to set `LMIC_ENABLE_DeviceTimeReq` to **1** in `config.h`, please find link below.
https://github.com/mcci-catena/arduino-lmic/blob/master/src/lmic/config.h#L172

terrillmoore, 2021-11-20:
@geronimoooooooo did this fix your problem? This is working for us, so this may not be a bug.

terrillmoore, 2021-11-20:
Now that TTN v3 is the standard, we should probably enable device time request by default in `config.h`.

cyberman54, 2021-12-28:
I can confirm that getting the time from TTN v3 network is working this way. And it's pretty precise, i see +/- 100ms.

terrillmoore, 2021-12-28:
@cyberman54 Yes, we've found the same thing.  It sometimes takes several queries to get the time, but when it comes in, it works well. I'll create an issue to track changing the default.

closed 2021-12-28 by terrillmoore