mcci-catena/arduino-lmic #822: Not working on EU433Mhz (Arduino UNO + Adafruit RF95W 433Mhz LoRa module)

jndqadar, 2021-11-24:
@terrillmoore 
Dear Mr. Terry Moore,

Thanks for your efforts and providing such a good library. I am trying to connect Arduino Uno with Adafruit RF95W LoRa module (EU433 Mhz), however it doesn't work on the said frequency plan. I am trying connecting from Genova, Italy and pretty sure this frequency plan can be supporting in this region. Can you please provide me your kind direction is fixing out this problem? Please refer me to the alternative solution if 433 Mhz is not sorted out yet in this library. 
Your support will be highly appreciated. 

Please have a loot at the snapshot  attached if you could provide me help in sorting out
[image]

cstratton, 2021-12-02:
Do you have your own 433 MHz gateway operating an connected to whatever server infrastructure you registered your node with?

It's extremely unlikely that anyone else is running a 433 MHz gateway on something like TTN within range of you, since 433 MHz is not a commonly used band for LoRaWAN in Europe, and even coverage of the usual 868 MHz band is far from total.

jndqadar, 2021-12-02:
> Do you have your own 433 MHz gateway operating and connected to whatever server infrastructure you registered your node with?
> 
> It's extremely unlikely that anyone else is running a 433 MHz gateway on something like TTN within range of you, since 433 MHz is not a commonly used band for LoRaWAN in Europe, and even coverage of the usual 868 MHz band is far from total.

jndqadar, 2021-12-02:
Hi, you are absolutely right, I tried a lot, but the 433 MHz single-channel gateway does not connect with TTN. I just wanna try it on my Pi terminal as it works on the pi terminal and shows payload and time stamp etc. Therefore, I am trying to connect the 433 MHz with a single-channel packet forwarder working on my pi terminal.

cstratton, 2021-12-02:
TTN explicitly prohibits single channel devices that pretend to be gateways

jndqadar, 2021-12-02:
So, is all the hardware I bought just useless? Or is there any other solution to utilize them? I bought 3 Adafruit RF95W (433 MHz) Lora modules with 2 RPi.

cstratton, 2021-12-02:
You can consider peer to peer usages, but yes, for LoRaWAN its pretty much useless

jndqadar, 2021-12-02:
Thank you for your kind help. Just one last question, can I utilize them with another network server I,e chipstack, thingspark or these are totally wasted in the EU region?

cstratton, 2021-12-02:
The first obstacle is that you'd need an actual LoRaWAN gateway based on a multichannel concentrator card.

The next is whatever regulatory restrictions apply to the 433 MHz band in your location.

jndqadar, 2021-12-02:
Thank you so much, you helped me a lot.