mcci-catena/arduino-lmic #827: EV_TXSTART EV_JOIN_TXCOMPETE:no JoinAccept

liuhengjing-sudo, 2021-12-10:
**Consider raising support questions on the forum first**

There is a discussion site, [forum.mcci.io](https://forum.mcci.io), which includes a [category focused on the Arduino LMIC](https://forum.mcci.io/c/device-software/arduino-lmic/5). Unless you're sure that your problem is a bug in the LMIC, it would be great if you can try for help there first. That will keep issues focused on work that needs to be done by the developers.

**Describe your question or issue**

Please give a clear and concise description of the problem you're facing and what you'd like help with.

**Environment**

This information is very important; it's hard to help without a complete set of answers.

- Version of LMIC being used. If using the latest github mainline, let us know, otherwise state the version.
- Version of Arduino IDE being used:
- Network provider (The Things Network, Swisscom, ChirpStack, etc.)
- Region (EU868, US915, etc.)
- Board (MCCI Catena, Adafruit Feather M0, Heltec Wi-Fi LoRa 32 v2, etc.)
- Radio (HopeRF, SX1276, etc.)

Hello

I am using lora shield v1.4 and arduino uno and au915 and using the ttn-otaa example
The port monitoring only show EV_TXSTART EV_JOIN_TXCOMPLETE :no join accept
And my applicant data in TTN is no data appear
My gateway is self-build single channel using rasp pi and lora shield v1.4 and there are no data appear on gateway in TTN

my pin mapping is 
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 5,
    .dio = {2, 3, LMIC_UNUSED_PIN},
};

Any help will be grateful
Thanks !

terrillmoore, 2021-12-10:
Hi,

Sorry you're having problems.

If you have two devices, please use the raw sketch to confirm radio tx/rx performance.

After that, try setting the clock error to a large tolerance, e.g. `LMIC_setClockError(MAX_CLOCK_ERROR * 20 / 1000)`.  Also, at the same time, in your project configuration file, add `#define LMIC_ENABLE_arbitrary_clock_error 1`. This often helps. 

If not, please share your project config file and test sketch.

Best regards,
--Terry

liuhengjing-sudo, 2021-12-11:
Thanks for your detailed and quick reply 
I am sorry I am very new in the Lora. 
After I put LMIC_setClockError(MAX_CLOCK_ERROR * 20 / 1000) before dosend() in ttn-otaa, and put #define LMIC_ENABLE_arbitrary_clock_error 1 in Arduino\libraries\arduino-lmic-master\project_config. I get this in my rasp pi terminal 
[image]
However I still get EV_TXSTART EV_JOIN_TXCOMPLETE :no join accept in arduino port monitoring. 
[image]

my sketch code is 
#include <lmic.h>
#include <hal/hal.h>
#include <SPI.h>

//
// For normal use, we require that you edit the sketch to replace FILLMEIN
// with values assigned by the TTN console. However, for regression tests,
// we want to be able to compile these scripts. The regression tests define
// COMPILE_REGRESSION_TEST, and in that case we define FILLMEIN to a non-
// working but innocuous value.
//
#ifdef COMPILE_REGRESSION_TEST
# define FILLMEIN 0
#else
# warning "You must replace the values marked FILLMEIN with real values from the TTN control panel!"
# define FILLMEIN (#dont edit this, edit the lines that use FILLMEIN)
#endif

// This EUI must be in little-endian format, so least-significant-byte
// first. When copying an EUI from ttnctl output, this means to reverse
// the bytes. For TTN issued EUIs the last bytes should be 0xD5, 0xB3,
// 0x70.
static const u1_t PROGMEM APPEUI[8]={0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};
void os_getArtEui (u1_t* buf) { memcpy_P(buf, APPEUI, 8);}

// This should also be in little endian format, see above.
static const u1_t PROGMEM DEVEUI[8]= {0x70, 0xB3, 0xD5, 0x7E, 0xD0, 0x04, 0x9C, 0x26};
void os_getDevEui (u1_t* buf) { memcpy_P(buf, DEVEUI, 8);}

// This key should be in big endian format (or, since it is not really a
// number but a block of memory, endianness does not really apply). In
// practice, a key taken from ttnctl can be copied as-is.
static const u1_t PROGMEM APPKEY[16] = {0xC2, 0x19, 0xBF, 0x2F, 0x4F, 0xC0, 0x87, 0x89, 0x6A, 0xDA, 0x59, 0xCA, 0x18, 0x8E, 0xFA, 0x07};
void os_getDevKey (u1_t* buf) {  memcpy_P(buf, APPKEY, 16);}

static uint8_t mydata[] = "Hello, world!";
static osjob_t sendjob;

// Schedule TX every this many seconds (might become longer due to duty
// cycle limitations).
const unsigned TX_INTERVAL = 60;

// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 5,
    .dio = {2, 3, LMIC_UNUSED_PIN},
};

void printHex2(unsigned v) {
    v &= 0xff;
    if (v < 16)
        Serial.print('0');
    Serial.print(v, HEX);
}

void onEvent (ev_t ev) {
    Serial.print(os_getTime());
    Serial.print(": ");
    switch(ev) {
        case EV_SCAN_TIMEOUT:
[... 138 lines cut ...]

    // LMIC init
    os_init();
    // Reset the MAC state. Session and pending data transfers will be discarded.
    LMIC_reset();

    // Use with Arduino Pro Mini ATmega328P 3.3V 8 MHz
    // Let LMIC compensate for +/- 1% clock error
    LMIC_setClockError(MAX_CLOCK_ERROR * 20 / 1000);
   
    // Start job (sending automatically starts OTAA too)
    do_send(&sendjob);
}

void loop() {
    os_runloop_once();
}

and my project config file is   in libraries 
[image]
[image]
[image]

the rasp pi code is come from [this link ](https://www.instructables.com/Use-Lora-Shield-and-RPi-to-Build-a-LoRaWAN-Gateway/) , which is compiled by sudo ./single-chanel-ptk-fwd
very appreciate!

terrillmoore, 2021-12-11:
A Join message is always sent at SF10. The "single channel gateway" is listening at SF7. This means that it won't hear the JoinRequest message (for non-gateway LoRa chips, the TX and RX SF *must* match (as must the frequency).

The LMIC can't easily be changed to use SF7 for JoinRequest, but perhaps the SingleChannel gateway software can be changed to always listen on SF10 rather than SF7.

Single channel gateways are very troublesome and are the **very hardest** way to get started with LoRa. I strongly suggest spending a little money (US $80) for a Things Network Indoor Gateway. Whoever introduced single-channel gateways did learners a great disservice, because it looks like an inexpensive way to start. But it also leads to frustration and gives a distorted view of the technology. (I'm not sure TTN really supports single-channel gateways, in fact.)

liuhengjing-sudo, 2021-12-11:
Thanks for your reply
After I put the gateway outdoor just the yard, the node is connected and send the uplink message slowly.
However after I want retry this, there is loop in Applicant live data:
Forward join accept message
Accept join request

The port monitoring in arduino is still no join accept
I try SF10 it does not work
And I try the ttn ABP,  with the DEVADRR NESKEY APPSKEY from this  end device in Applicant which gives me failure message in port monitoring.
I am still adjust the distance between gateway and the node,  sometimes the applicant shoews that the joint sever failed, I  am not sure what is the problem right now 

But Thanks again!

MrAnnihilaToR, 2022-12-15:
OK guys, I had the same problem, so let me tell you how I fixed it : I went to lmic.project.config , and there I disabled all the region settings and just left #define LMIC_ENABLE_arbitrary_clock_error 1 and #define CFG_sx1276_radio 1 , then you go to your basic settings .ini file -> in my case is platformio.ini and there you must add your region like this :

 build_flags =
 -D CFG_eu868=1
 
 This worked for me, let's hope it will help somebody too. :)