mcci-catena/arduino-lmic #828: Connecting Catena to Helium Network

mhmayyan, 2021-12-11:
Previously, I was able to transmit data to the helium network. I am still using the same device but it has been off for months. Now, I am unable to receive data because the device keeps trying to join. I also tried to add a new device but it also got stuck at the joining request as shown below.

```
EV_JOINING
NwkID:   00000000   DevAddr: 00000000
EV_TXSTART
EV_JOIN_TXCOMPLETE
EV_TXSTART
EV_JOIN_TXCOMPLETE
```

It keeps repeating this forever (EV_TXSTART and EV_JOIN_TXCOMPLETE). I am able to see the join requests on the Helium console but it doesn't send the payload. I selected Helium for the LoRaWAN network and used the default for the subband. 
I tried to add `LMIC_setClockError(MAX_CLOCK_ERROR * 20 / 1000)` after initializing `Catena::LoRaWAN` and I added `#define LMIC_ENABLE_arbitrary_clock_error 1` in the project configuration file. This did not help.

```
system version 
Board: Catena 4612
Platform-Version: 0.21.2
Arduino-LoRaWAN-Version: 0.9.1
Arduino-LMIC-Version: 4.1.0
MCCIADK-Version: 0.2.2
MCCI-Arduino-BSP-Version: 3.0.4
```

Please help. 

Thanks.

terrillmoore, 2021-12-11:
My suggestion is that you first make sure the same device will connect to TTNv3.  Then go back to Helium, set the subband to channels 8~15/65 (via the IDE) and then make sure that you have the option *disabled* in the Helium channel to extend the JoinAccept with the CFlist. The Helium network has a fair number of differences related to the LoRaWAN standard, and adding CFlist won't work with the LMIC (not our bug).

Regardless of compatibility issues, if Helium is hearing the device, you'll see uplinks for Join Requests in the console. 

If it's not, make sure your Helium gateway is working. Cycle power if needed.

If you're still having problems, please set the output/build directory for the Arduino IDE (so you can find the output), and then capture and post the `build.options.json` that will show up in the output directory after a build.

Best regards,
--Terry

mhmayyan, 2021-12-12:
I sorted out this problem by reconfiguring my gateway, which is RAK7244P. I just followed [these instructions](https://github.com/RAKWireless/RAK2245_for_helium).

Thanks, Terry.

mrx23dot, 2022-02-21:
Please close the ticket then.

closed 2022-02-21 by mhmayyan