mcci-catena/arduino-lmic #830: Wisol LOM204 (stm32L071 + SX1276

tcpipchip, 2021-12-11:
**Describe the bug**

After changed the JSON to the new STM32 Arduino GITHUB,  the RADIO enter in infinite loop on  "while" sentence

`    // set the tcxo input, if needed
    if (hal_queryUsingTcxo())
        writeReg(RegTcxo, readReg(RegTcxo) | RegTcxo_TcxoInputOn);

    // seed 15-byte randomness via noise rssi
    rxlora(RXMODE_RSSI);
    //while( (readReg(RegOpMode) & OPMODE_MASK) != OPMODE_RX ); // continuous rx
    for(int i=1; i<16; i++) {
        for(int j=0; j<8; j++) {
            u1_t b; // wait for two non-identical subsequent least-significant bits
            //while( (b = readReg(LORARegRssiWideband) & 0x01) == (readReg(LORARegRssiWideband) & 0x01) );
            randbuf[i] = (randbuf[i] << 1) | b;
        }
    }
    randbuf[0] = 16; // set initial index`

Wireless is tricky; if in doubt, start with a support request, rather than a bug report.

**Environment**

- Version of LMIC being used. If using the latest GitHub mainline, let us know, otherwise state the version.
- Version of Arduino IDE being used.
1.8.16

- Network provider (The Things Network, Swisscom, ChirpStack, etc.)
TTN

- Region (EU868, US915, etc.)

AU915

- Board (MCCI Catena, Adafruit Feather M0, Heltec, etc.)

LOM204 (STM32L071 + SX1276)

- CPU (AVR, STM32L0, etc.)

STM32L071

- Radio (HopeRF, SX1276, etc.)

SX1276

**To Reproduce**

I dont known if you know LOM204 (WISOL)

**Expected behavior**

Enter in LOOP.

8 months was working...but, with the old STM32 repository and a old LMIC lib,  
sorry, i dont registered wich version were, but it was working 

**Screenshots**

I modified to show the value read.

```
  // seed 15-byte randomness via noise rssi
    rxlora(RXMODE_RSSI);
    while( (readReg(RegOpMode) & OPMODE_MASK) != OPMODE_RX )
		core_debug("%d\r\n",readReg(RegOpMode));
```

**Additional context**

Add any other context about the problem here.

[image]
10001000

tcpipchip, 2022-01-20:
all Working Now!
I had to include enable the TXCO and SWITCH PIN

see the BLOG

https://lom204-cli-wisol.blogspot.com/2022/01/importante-opcao-nao-oficial-da-wisol.html

terrillmoore, 2022-01-20:
That's great, congratulations!  Closing the ticket...

terrillmoore, 2022-01-20:
BTW, feel free to send a PR to add the configuration to the built-in config if you'd like.

closed 2022-01-20 by terrillmoore

tcpipchip, 2022-01-20:
Hi @terrillmoore 

how can we do that ? 

It’s a STM32 sdk config

terrillmoore, 2022-01-20:
Take a look at the files src/hal/getpinmap*.cpp, and at the corresponding header file. Ask questions if anything isn't clear.

tcpipchip, 2022-01-21:
@terrillmoore 
I created
`/*

Module:  getpinmap_wisol_lom204.cpp

Function:
        Arduino-LMIC C++ HAL pinmaps for the WISOL LOM204

Copyright & License:
        See accompanying LICENSE file.

Author:
        Miguel Alexandre Wisintainer

*/

#if defined(ARDUINO_WISOL_LOM204)

#include <arduino_lmic_hal_boards.h>
#include <Arduino.h>

#include "../lmic/oslmic.h"

namespace Arduino_LMIC { 

    class HalConfiguration_wisol_lom204_t : public HalConfiguration_t
            {
    public:
            enum DIGITAL_PINS : uint8_t
                    {
                    PIN_SX1276_NSS = RADIO_NSS,
                    PIN_SX1276_NRESET = RADIO_RESET,
                    PIN_SX1276_DIO0 = RADIO_DIO_0,
                    PIN_SX1276_DIO1 = RADIO_DIO_1,
                    PIN_SX1276_DIO2 = RADIO_DIO_2,
                    PIN_SX1276_RXTX = RADIO_ANT_SWITCH_HF,
                    };

        virtual bool queryUsingTcxo(void) override { return false; };
            };
    // save some typing by bringing the pin numbers into scope
    static HalConfiguration_wisol_lom204_t myConfig;

    static const HalPinmap_t myPinmap =
            {
            .nss = HalConfiguration_wisol_lom204_t::PIN_SX1276_NSS,
            .rxtx = HalConfiguration_wisol_lom204_t::PIN_SX1276_RXTX, 
            .rst = HalConfiguration_wisol_lom204_t::PIN_SX1276_NRESET,

            .dio = {HalConfiguration_wisol_lom204_t::PIN_SX1276_DIO0, 
                    HalConfiguration_wisol_lom204_t::PIN_SX1276_DIO1, 
                    HalConfiguration_wisol_lom204_t::PIN_SX1276_DIO2, 
                }
            };

    const HalPinmap_t *GetPinmap_wisol_lom204(void)
            {
            return &myPinmap;
            }

}; // namespace Arduino_LMIC

#endif /* defined(ARDUINO_WISOL_LOM204) */`

tcpipchip, 2022-01-21:
what more have i to do ?

`/*******************************************************************************
 * Copyright (c) 2015 Thomas Telkamp and Matthijs Kooijman
 * Copyright (c) 2018 Terry Moore, MCCI
 *
 * Permission is hereby granted, free of charge, to anyone
 * obtaining a copy of this document and accompanying files,
 * to do whatever they want with them without any restriction,
 * including, but not limited to, copying, modification and redistribution.
 * NO WARRANTY OF ANY KIND IS PROVIDED.
 *
 * This example sends a valid LoRaWAN packet with payload "Hello,
 * world!", using frequency and encryption settings matching those of
 * the The Things Network.
 *
 * This uses OTAA (Over-the-air activation), where where a DevEUI and
 * application key is configured, which are used in an over-the-air
 * activation procedure where a DevAddr and session keys are
 * assigned/generated for use with all further communication.
 *
 * Note: LoRaWAN per sub-band duty-cycle limitation is enforced (1% in
 * g1, 0.1% in g2), but not the TTN fair usage policy (which is probably
 * violated by this sketch when left running for longer)!

 * To use this sketch, first register your application and device with
 * the things network, to set or generate an AppEUI, DevEUI and AppKey.
 * Multiple devices can use the same AppEUI, but each device has its own
 * DevEUI and AppKey.
 *
 * Do not forget to define the radio type correctly in
 * arduino-lmic/project_config/lmic_project_config.h or from your BOARDS.txt.
 *
 *******************************************************************************/

#include <lmic.h>
#include <hal/hal.h>
#include <SPI.h>

HardwareSerial Serial1(PB4, PB3);

//
// For normal use, we require that you edit the sketch to replace FILLMEIN
// with values assigned by the TTN console. However, for regression tests,
// we want to be able to compile these scripts. The regression tests define
// COMPILE_REGRESSION_TEST, and in that case we define FILLMEIN to a non-
// working but innocuous value.
//
#ifdef COMPILE_REGRESSION_TEST
# define FILLMEIN 0
#else
# warning "You must replace the values marked FILLMEIN with real values from the TTN control panel!"
# define FILLMEIN (#dont edit this, edit the lines that use FILLMEIN)
#endif

// This EUI must be in little-endian format, so least-significant-byte
// first. When copying an EUI from ttnctl output, this means to reverse
// the bytes. For TTN issued EUIs the last bytes should be 0xD5, 0xB3,
// 0x70.

static const u1_t PROGMEM APPEUI[8]={ FILLMEIN };

void os_getArtEui (u1_t* buf) { memcpy_P(buf, APPEUI, 8);}

// This should also be in little endian format, see above.
static const u1_t PROGMEM DEVEUI[8]={ FILLMEIN };
void os_getDevEui (u1_t* buf) { memcpy_P(buf, DEVEUI, 8);}

// This key should be in big endian format (or, since it is not really a
// number but a block of memory, endianness does not really apply). In
[... 161 lines cut ...]
    //LOM204
    Serial.begin(9600);
    Serial.println(F("Starting"));

    #ifdef VCC_ENABLE
    // For Pinoccio Scout boards
    pinMode(VCC_ENABLE, OUTPUT);
    digitalWrite(VCC_ENABLE, HIGH);
    delay(1000);
    #endif

    // LMIC init
    os_init();
    // Reset the MAC state. Session and pending data transfers will be discarded.
    LMIC_reset();
    LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);
    LMIC_selectSubBand(1); //FSB2 start in 0 //TTN
    // Start job (sending automatically starts OTAA too)
    //LMIC_startJoining();
    do_send(&sendjob);
}

void loop() {
    os_runloop_once();
}`

tcpipchip, 2022-01-21:
`/*******************************************************************************
 * Copyright (c) 2015 Thomas Telkamp and Matthijs Kooijman
 * Copyright (c) 2018 Terry Moore, MCCI
 *
 * Permission is hereby granted, free of charge, to anyone
 * obtaining a copy of this document and accompanying files,
 * to do whatever they want with them without any restriction,
 * including, but not limited to, copying, modification and redistribution.
 * NO WARRANTY OF ANY KIND IS PROVIDED.
 *
 * This example sends a valid LoRaWAN packet with payload "Hello,
 * world!", using frequency and encryption settings matching those of
 * the The Things Network.
 *
 * This uses OTAA (Over-the-air activation), where where a DevEUI and
 * application key is configured, which are used in an over-the-air
 * activation procedure where a DevAddr and session keys are
 * assigned/generated for use with all further communication.
 *
 * Note: LoRaWAN per sub-band duty-cycle limitation is enforced (1% in
 * g1, 0.1% in g2), but not the TTN fair usage policy (which is probably
 * violated by this sketch when left running for longer)!

 * To use this sketch, first register your application and device with
 * the things network, to set or generate an AppEUI, DevEUI and AppKey.
 * Multiple devices can use the same AppEUI, but each device has its own
 * DevEUI and AppKey.
 *
 * Do not forget to define the radio type correctly in
 * arduino-lmic/project_config/lmic_project_config.h or from your BOARDS.txt.
 *
 *******************************************************************************/

#include <lmic.h>
#include <hal/hal.h>
#include <SPI.h>

HardwareSerial Serial1(PB4, PB3);

//
// For normal use, we require that you edit the sketch to replace FILLMEIN
// with values assigned by the TTN console. However, for regression tests,
// we want to be able to compile these scripts. The regression tests define
// COMPILE_REGRESSION_TEST, and in that case we define FILLMEIN to a non-
// working but innocuous value.
//
#ifdef COMPILE_REGRESSION_TEST
# define FILLMEIN 0
#else
# warning "You must replace the values marked FILLMEIN with real values from the TTN control panel!"
# define FILLMEIN (#dont edit this, edit the lines that use FILLMEIN)
#endif

// This EUI must be in little-endian format, so least-significant-byte
// first. When copying an EUI from ttnctl output, this means to reverse
// the bytes. For TTN issued EUIs the last bytes should be 0xD5, 0xB3,
// 0x70.

static const u1_t PROGMEM APPEUI[8]={ FILLMEIN };

void os_getArtEui (u1_t* buf) { memcpy_P(buf, APPEUI, 8);}

// This should also be in little endian format, see above.
static const u1_t PROGMEM DEVEUI[8]={ FILLMEIN };
void os_getDevEui (u1_t* buf) { memcpy_P(buf, DEVEUI, 8);}

// This key should be in big endian format (or, since it is not really a
// number but a block of memory, endianness does not really apply). In
// practice, a key taken from ttnctl can be copied as-is.
static const u1_t PROGMEM APPKEY[16] = { FILLMEIN };
[... 159 lines cut ...]
    //LOM204
    Serial.begin(9600);
    Serial.println(F("Starting"));

    #ifdef VCC_ENABLE
    // For Pinoccio Scout boards
    pinMode(VCC_ENABLE, OUTPUT);
    digitalWrite(VCC_ENABLE, HIGH);
    delay(1000);
    #endif

    // LMIC init
    os_init();
    // Reset the MAC state. Session and pending data transfers will be discarded.
    LMIC_reset();
    LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);
    LMIC_selectSubBand(1); //FSB2 start in 0 //TTN
    // Start job (sending automatically starts OTAA too)
    //LMIC_startJoining();
    do_send(&sendjob);
}

void loop() {
    os_runloop_once();
}`

tcpipchip, 2022-01-21:
[ino.txt](https://github.com/mcci-catena/arduino-lmic/files/7909747/ino.txt)