mcci-catena/arduino-lmic #834: Configuring MISO, MOSI and SCK SPI pins?

sw-dev-code, 2021-12-21:
I would like to know how to configure the MISO, MOSI, SCK and SS pins for SPI in the case when I'm using the custom board?

From hal.cpp I can see that the library uses default pins:

```
static void hal_spi_init () {
    SPI.begin();
}
```

terrillmoore, 2021-12-21:
Look at the mechanisms used by the various halconfig entries. No matter what it seems, the HAL really does use the information in the tables. (See, for example, `src/hal/getpinmap_heltec_lora32.cpp`.)

sw-dev-code, 2021-12-21:
I'm already using the pin maps like in the following code, but there is no option to set MISO, MOSI and SCK pins. How can I change that without modifying the library?

```
const cMyLoRaWAN::lmic_pinmap myPinMap = {
     .nss = LORA_SS,
     .rxtx = cMyLoRaWAN::lmic_pinmap::LMIC_UNUSED_PIN,
     .rst = LORA_RESET,
     .dio = {LORA_IRQ , LORA_MOSI, cMyLoRaWAN::lmic_pinmap::LMIC_UNUSED_PIN },
    //  .rxtx_rx_active = 0,
    //  .rssi_cal = 0,
    //  .spi_freq = 8000000,
};
```

terrillmoore, 2021-12-21:
Ah, yes, sorry. The library uses the global object `SPI`, and that can't be changed without a lot of work. So you have to create the global `SPI` and set up the pins there before starting the libarry. There's no way at present to change that. Perhaps in version 5.

sw-dev-code, 2021-12-21:
@terrillmoore Thank you. I would be very grateful if in version 5 we can change those pins outside the library. I've already seen that some people would also benefit from that feature.

d-a-v, 2021-12-22:
Changing constant parameters such as buffer size or pinouts from outside the library would nearly always increase code, and this is sometimes harmful on small targets like at328p (which otherwise works great).
Usually C preprocessor global `#define` are used for this purpose.
The Arduino IDE environment does not allow users to use such \#defines globally.
You can switch to another environment which allows to do that.

About Arduino IDE, there have been many attempts to enhance user experience for this purpose, with no luck so far.
I will only [show one](https://github.com/arduino/arduino-cli/pull/1524), it includes direct and indirect links to many other past attempts.
Why this particular one ?
Because it is my 3rd proposal and I believe I made it also because of the same frustration that you have with arduino-lmic @sw-dev-code .

sw-dev-code, 2021-12-22:
@d-a-v Thank you so much for your explanation and help. I'm using the PlatformIO IDE. Is there any solution for that IDE?

d-a-v, 2021-12-22:
You can use or add [global defines with PIO](https://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-flags).
In order for users to be able to change defaults, libraries should use defines like this:

```c
#define THIS_LIB_SOME_PARAMETER_DEFAULT_MS   5 // ms
#ifndef THIS_LIB_SOME_PARAMETER_MS
#define THIS_LIB_SOME_PARAMETER_MS THIS_LIB_SOME_PARAMETER_DEFAULT_MS
#endif
```

You would have to add this in your PIO configuration file.
```c
-D THIS_LIB_SOME_PARAMETER_MS=10
```

terrillmoore, 2021-12-22:
The library already supports this, but in a slightly different way. see instructions on using with PlatformIO or the test cases.

sw-dev-code, 2021-12-22:
@terrillmoore I know that, but still, I think it is impossible to set MISO, MOSI and SCK pins this way in PlatformIO. 
If it is possible please tell me how.

cyberman54, 2021-12-28:
@sw-dev-code the way to create a global SPI, and init LMIC after, would work like this example shows:

```
class MyHalConfig_t : public Arduino_LMIC::HalConfiguration_t {

public:
  MyHalConfig_t(){};

  // set SPI pins to board configuration, pins may come from pins_arduino.h
  virtual void begin(void) override {
    SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_CS);
  }
};

static MyHalConfig_t myHalConfig{};

// LMIC pin mapping for Hope RFM95 / HPDtek HPD13A transceivers
static const lmic_pinmap myPinmap = {
    .nss = LORA_CS,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = LORA_RST == NOT_A_PIN ? LMIC_UNUSED_PIN : LORA_RST,
    .dio = {LORA_IRQ, LORA_IO1,
            LORA_IO2 == NOT_A_PIN ? LMIC_UNUSED_PIN : LORA_IO2},
    .rxtx_rx_active = LMIC_UNUSED_PIN,
    .rssi_cal = 10,
    .spi_freq = 8000000, // 8MHz
    .pConfig = &myHalConfig};

// setup LMIC stack
  os_init_ex(&myPinmap); // initialize lmic run-time environment

```