mcci-catena/arduino-lmic #838: LinkCheckReq Mac command during compliance mode

GitTibbe, 2021-12-28:
**Describe the bug**
When I runt the LoRaWAN LCTT pre-certification tool there TP_A_EU868_ED_MAC_BV_019 that is supposed to trigger a LinkCheckReq command from the end device(DUT). The test fails due to "MAC command not received because FPort field is not set to 0 and FOpts field is empty". Is this supported or does it need to be implemented in the lmic_compliance.c code?

```
00:00:39.288 	=========================================================
00:00:39.294 	=========================================================
00:00:39.296 	Start Test Step: LinkCheckReq Mac command
00:00:39.297 	Step 01
00:00:39.299 	Test Trigger LinkCheckReq process starts
00:00:42.416 	PULL_DATA: 02 1c0f 02 b827ebfffe88f94b
00:00:42.418 	PULL_ACK: 02 1c0f 04
00:00:48.228 	PUSH_ACK: 02 28bd 01
00:00:48.231 	PUSH_DATA: Protocol version = 02, Token = 28bd, Identifier = 00, JSON = {'rxpk': [{'tmst': 2334839355, 'chan': 0, 'rfch': 1, 'freq': 868.1, 'stat': 1, 'modu': 'LORA', 'datr': 'SF7BW125', 'codr': '4/5', 'lsnr': 4.2, 'rssi': -58, 'size': 15, 'data': 'QBEiM0SAAgDg6LPEFije'}]}
00:00:48.233 	DATA: (MHDR: 40, MACPayload: (FHDR: (DevAddr: 11223344, FCtrl: 80, FCnt: 0002, FOpts: ), FPort: e0, FRMPayload: 0000, MIC: c41628de)
00:00:48.236 	Server <-------------------- DUT
00:00:48.237 	PHYPayload: (MHDR: 60, MACPayload: (FHDR: (DevAddr: 11223344, FCtrl: 00, FCnt: 0002, FOpts: ), FPort: e0, FRMPayload: 05), MIC: 7e330e9e)
00:00:48.240 	PULL_RESP: Protocol version = 02, Token = 5e0e, Identifier = 03, JSON = {'txpk': {'imme': False, 'tmst': 2336839355, 'rfch': 0, 'powe': 14, 'ipol': True, 'datr': 'SF12BW125', 'freq': 869.525, 'modu': 'LORA', 'codr': '4/5', 'prea': 8, 'ncrc': True, 'size': 14, 'data': 'YBEiM0QAAgDgon4zDp4='}}
00:00:48.244 	Server --------------------> DUT
00:00:48.246 	Trigger LinkCheckReq communication tested successfully
00:00:48.248 	Step 02-03
00:00:48.249 	Wait a MAC command from the DUT process starts
00:00:52.431 	PULL_DATA: 02 bc86 02 b827ebfffe88f94b
00:00:52.434 	PULL_ACK: 02 bc86 04
00:00:58.314 	PUSH_ACK: 02 4e5c 01
00:00:58.317 	PUSH_DATA: Protocol version = 02, Token = 4e5c, Identifier = 00, JSON = {'rxpk': [{'tmst': 2344839091, 'chan': 2, 'rfch': 1, 'freq': 868.5, 'stat': 1, 'modu': 'LORA', 'datr': 'SF7BW125', 'codr': '4/5', 'lsnr': 8.0, 'rssi': -76, 'size': 15, 'data': 'QBEiM0SAAwDgk7fDebMP'}]}
00:00:58.318 	DATA: (MHDR: 40, MACPayload: (FHDR: (DevAddr: 11223344, FCtrl: 80, FCnt: 0003, FOpts: ), FPort: e0, FRMPayload: 0001, MIC: c379b30f)
00:00:58.320 	Server <-------------------- DUT
00:00:58.322 	MAC command not received because FPort field is not set to 0 and FOpts field is empty
00:00:58.323 	MAC Command received is not correct
00:00:58.325 	End Test Step: LinkCheckReq Mac command
00:00:58.326 	Partial Verdict: FAIL
00:00:58.331 	=========================================================
00:00:58.338 	=========================================================
```

**Environment**

LMIC library 4.1.1, LoRaWAN 1.0.2 version

**Expected behavior**

Should send an uplink LinkCheckReq Mac command from the DUT.

terrillmoore, 2021-12-28:
That command is currently not implemented. Duplicate of #825, I think.

GitTibbe, 2021-12-30:
By adding the following function under case LORAWAN_COMPLIANCE_CMD_LINK, the compliance test TP_A_EU868_ED_MAC_BV_019 passes. However it is then only used during test-mode mode and never in normal mode. 

```
static void evLinkCheckReqCommand() {

    uint8_t cmd = MCMD_LinkCheckReq;
    if (LMIC.pendMacPiggyback) {
        // put in pendMacData
        if (LMIC.pendMacLen < sizeof(LMIC.pendMacData)) {
            LMIC.pendMacData[LMIC.pendMacLen++] = cmd;
        } else {
            return;
        }
    } else {
        // put in pendTxData
        if (LMIC.pendMacLen < sizeof(LMIC.pendTxData)) {
            LMIC.pendTxData[LMIC.pendMacLen++] = cmd;
        } else {
            return;
        }
    }

    fsmEvalDeferred();
}

```
  call it in:
```
case LORAWAN_COMPLIANCE_CMD_LINK: {
            evLinkCheckReqCommand(); // we are required to initiate a Link
            break;
        }
```