mcci-catena/arduino-lmic #841: rx1DrOffset > 3 in EU868 not working

GitTibbe, 2021-12-29:
**Describe the bug**

I noticed that the MAC command MCMD_RXParamSetupReq does not seem to be adapted for EU868 specs, but only for US like specs. 

The following lines in the case makes the LoRaWAN LCTT test case TP_A_EU868_ED_MAC_BV_015_A (RX1 Receive window test Part 1) fail when the requested data rate offset for Rx1 is 4 or 5. I suggest changing the number 3 to a query function for each region for the maximum supported RX1 data rate range. For the EU868 test, if I change it from 3 to 5 in the if statement, the test passes, but that would not be good for the other regions neither. 

I suggest to change this: 
```
if (rx1DrOffset <= 3)
    LMIC.dn2Ans |= MCMD_RXParamSetupAns_RX1DrOffsetAck;
```
to e.g.:

```
if (rx1DrOffset <= LMIC_queryMaxRx1DrOffset())
    LMIC.dn2Ans |= MCMD_RXParamSetupAns_RX1DrOffsetAck;
```
Not sure if there are more things to change or not. 

The data rate range for RX1 in each region are specified here: https://lora-alliance.org/wp-content/uploads/2020/11/rp_2-1.0.1.pdf#page=27.

Maybe there are more places where numbers like this are only suitable for a certain region? 

**Environment**

LMIC 4.1.1
LCTT 3.4.0
EU868 region

**To Reproduce**

Do a MAC command for rx1DrOffset value of 4 or 5.

terrillmoore, 2021-12-29:
Thanks, I'll take a look. I depended on the v1.0.2 test scripts, which probably didn't test this.

> Maybe there are more places where numbers like this are only suitable for a certain region?

Possibly. This is in lmic.c, and I simply was in too much of a hurry when doing the conversion, and overlooked it here. Possibly elsewhere; but most likely in the MAC command parsing, because that's 99% new code, and was only tested with the 1.0.2 compliance scripts, which are different than the LCTT code.

Bear in mind, by the way, that FSK receive is not stable for me. I don't think it ever has been fully stable; but it's certainly is not stable in 4.1.1.

terrillmoore, 2022-02-19:
I went back to the LoRaWAN 1.0.3 regional spec (since we implement 1.0.3). The same problem exists.

Only US is limited to values 0..3. Most are limited to 0..5.

However, AS923 and IN865 allow values 6 and 7, which are *negative* offsets (so if uplink is DR0, downlink will be DR1 or DR2).

For AS923, it's even more complex:
[image]

So we not only need the routine to allow the values to be set, but we need to check where the actual Rx1DR is calculated. In fact, those look correct; so the only problem is the mac downlink check.