mcci-catena/arduino-lmic #842: How to disable FSK operations?

GitTibbe, 2021-12-30:
Due to FSK operations not being fully stable and not being mandatory for LoRaWAN, I would like to disable it to prevent any potential problems. 

What is the best way of disabling it? 

Using LMIC 4.1.1 EU868.

terrillmoore, 2021-12-30:
Good question. 

For testing, you should look at `LMICeu868_validDR()`:
https://github.com/mcci-catena/arduino-lmic/blob/4342770945ee2887c14969cec669861b387947ee/src/lmic/lmic_eu868.c#L52-L58

You could add:
```c
// don't allow FSK
if (dr == EU868_DR_FSK && LMICbandplan_disableFSK())
    return 0;
```

This doesn't get rid of the FSK support code, but it would let you continue. A similar change in `setDrTxpow` would make sure that client apps that respect the APIs would also feel this restriction:

https://github.com/mcci-catena/arduino-lmic/blob/4342770945ee2887c14969cec669861b387947ee/src/lmic/lmic.c#L415-L438

Change line 431 to:

```c
if( LMIC.datarate != dr && LMICbandplan_validDR(dr) ) {
```

It would, of course, be better to toss out the FSK code if not enabled... but that's a much more difficult change.

GitTibbe, 2022-01-07:
Thanks, that worked!

terrillmoore, 2022-01-07:
Great.

> Should also be the same way for SF7BW250 as its optional too

Let me mull this over for a day or two. (I'm almost completely single-handed here at the moment, due to a combination of the general situation we're all suffering through, but also due to some family situations that have cropped up. So I'm having to do inventory, shipping, all the physical stuff at the office. Feel free to ping me if I don't come back. Probably just another `if` clause after the one we added, checking for SF7BW250.. in the right way... And then look at the other regions for similar things.