mcci-catena/arduino-lmic #848: Always failure in radio.c 801

mnett0, 2022-01-30:
Hello, 

I have a problem when assembling a RFM95W module with an ESP32 (TTGO T1 v1.3) with the example code helium-oota. 

I have constantly in the serial monitor the following error: 
```
Starting
Packet queued
315615: EV_JOINING
537265: EV_TXSTART
FAILURE 
.pio/libdeps/ttgo-t1/MCCI LoRaWAN LMIC library/src/lmic/radio.c:801
```

Environment : 

- Version of LMIC : v4.1.1
- I use VSCode with PlatformIO IDE v2.4.0 (I have the same error with the Arduino IDE)
- Network provider : Helium
- Region : France in Europe with EU868
- Board : TTGO T1 v1.3
- Radio RFM95W(SX1276)
- Espressif 32 v3.5.0 

Schema : 

- ESP32 ↔ RFM95W
- MISO → 19
- MOSI → 23
- RST → 17 
- CS → 5
- SCK → 18
- DIO2 → 34
- DIO1 → 35
- DIO0 → 26

And I added this piece of code to the example: 
```
// SCK, MISO, MOSI, SS
SPI.begin(18, 19, 23, 5);

lmic_pinmap lmic_pins = {
  .nss = 5,
  .rxtx = LMIC_UNUSED_PIN,
  .rst = 17,
  .dio = {26, 35, 34},
};
 
const lmic_pinmap *pPinMap = &lmic_pins;
```

I tried the wiring also in HSPI same thing, I reversed NSS and SCK, nothing, I tried with a classic ESP32, same thing. On a classic Arduino board everything works fine. 

Anybody have an idea of the problem?

gmanrc, 2022-01-30:
Do you have anything else on the SPI line? I've recently been experiencing this failure when I have an MCP3008 ADC on the SPI line as well. Have not been able to make the two work nicely together.

mnett0, 2022-01-30:
I don't have anything connected on the VSPI, on the HSPI there is the micro-sd reader but I don't use it for my tests, and on a classic esp32 I also have this problem.

TTGO T1 schematic: https://github.com/LilyGO/ESP32-TTGO-T1/blob/master/T1_V1.3.pdf

mnett0, 2022-05-20:
@terrillmoore any idea?

mnett0, 2022-05-21:
Same with [XIAO BLE Sense](https://wiki.seeedstudio.com/XIAO_BLE/)