mcci-catena/arduino-lmic #856: Lora Discovery board L072CZ stuck in os_init()

arihantdaga, 2022-02-20:
**Describe your question or issue**

I am using Lora Discovery board L072CZ. Using the example helium-otaa. I used with platformio. 
I Gave correct EUIs and AppKey, when running the program it seems that the program is never crossing `os_init_ex(pPinMap);` line. (I kept serial prints to verify). On further debugging, in the function os_init_ex, the program is getting stuck at radio_init. 

Could i be doing something wrong ? 
Later i also checked pin configutaions. 
and found this in the `getpinmap_disco_l072cs_lrwan1.cpp` file

```C++
 enum DIGITAL_PINS : uint8_t
                     {
                     PIN_SX1276_NSS = 37,
                    PIN_SX1276_NRESET = 33,
                     PIN_SX1276_DIO0 = 38,
                     PIN_SX1276_DIO1 = 39,
                     PIN_SX1276_DIO2 = 40,
                     PIN_SX1276_RXTX = 21,
                     };
```

But on checking the doc `https://os.mbed.com/platforms/ST-Discovery-LRWAN1/ ` and https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32L0/TARGET_STM32L072xZ/TARGET_DISCO_L072CZ_LRWAN1/PeripheralPins.c
I found pins should have been different, so accoridingly i also changed the pins and updated pins looked like this - 
```C++
enum DIGITAL_PINS : uint8_t
                    {
                    PIN_SX1276_NSS = 35, // PA15
                    PIN_SX1276_NRESET = 31, // PC0
                    PIN_SX1276_DIO0 = 36, //PB4 
                    PIN_SX1276_DIO1 = 37, //PB1 
                    PIN_SX1276_DIO2 = 38, //PB0
                    PIN_SX1276_RXTX = 21, // PA1
                    };
```

But even after doing these changes, the program still remain stuck in radio_init

Could i be doing something wrong, can someone point me in right direction. 

**Environment**

This information is very important; it's hard to help without a complete set of answers.

- Version of LMIC being used - Latest release from platformio
- Network provider Helium
- Region - IN866
- Board (MCCI Catena, Adafruit Feather M0, Heltec Wi-Fi LoRa 32 v2, etc.) ST-Discovery Lora - https://os.mbed.com/platforms/ST-Discovery-LRWAN1/
- Radio SX1276

My PIO COnfig
```
platform = ststm32@15.2.0
board = disco_l072cz_lrwan1
framework = arduino
upload_protocol = stlink
build_flags = 
        -D ARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS
        -D CFG_in866=1
        -D CFG_sx1276_radio=1
        -D ARDUINO_DISCO_L072CZ_LRWAN1=1
```

chansheunglong, 2022-02-24:
Have you enabled all debug messages? If yes please share them so that we can take a look

arihantdaga, 2022-02-24:
I am not sure how to enable debug messages, can you give me a hint on how to do that ? What are the debug flags ?

chansheunglong, 2022-02-24:
> I am not sure how to enable debug messages, can you give me a hint on how to do that ? What are the debug flags ?

In "project_config" folder, open "lmic_project_config.h", add "#define LMIC_DEBUG_LEVEL 2"

> **Changing debug output**
> #define LMIC_PRINTF_TO SerialLikeObject
> 
> This variable should be set to the name of a Serial-like object, used for printing messages. If not defined, Serial is assumed.
> 
> **Getting debug from the RF library**
> #define LMIC_DEBUG_LEVEL number /* 0, 1, or 2 */
> 
> This variable determines the amount of debug output to be produced by the library. The default is 0.
> 
> If LMIC_DEBUG_LEVEL is zero, no output is produced. If 1, limited output is produced. If 2, more extensive output is produced. If non-zero, printf() is used, and the Arduino environment must be configured to support it, otherwise the sketch will crash at runtime.

But since I use Arduino IDE, not sure if the file location is the same for PIO or not.
I think PIO use "build_flags" ?

arihantdaga, 2022-02-24:
ok, i'll do

terrillmoore, 2022-02-24:
Please avoid using debug prints unless there is a good reason to do so.

**They break the operation of the LMIC.**

They're only good for finding logic errors. I intend to remove them from the next major release, because they are a trap for the inexperienced users; I'll replace them with an expansion of the existing logging system. 

The debug prints in this area (for os_init) ware not very helpful, in any case.

In this case, you have a STM32, and I think the Disco board even includes an STLINK (indeed, I see you are using STLINK for upload). This means you can use GDB (or some other debugger) to single step through the code and find out what is going wrong.  Please do that! It is quite easy to figure out what is wrong if you single step through the code and find out where it's looping and then post that code snipped. You can even run at full speed, and break in to find the hang.  PlatformIO has good support for this, the payback really will reward the investigation.