mcci-catena/arduino-lmic #858: Event: LMIC Error: LMIC_ERROR_TX_BUSY after 5 uplink packets

sw-dev-code, 2022-02-23:
I'm using the library on ESP32 and I'm storing the LMIC data to avoid joining after each deep sleep.
The problem I have is that after 5 uplink packets I'm getting Event: LMIC Error: LMIC_ERROR_TX_BUSY when I try to send data.

I tried with two different intervals between packets and the error is the same. Are there some rules or parameters I'm missing?

terrillmoore, 2022-02-23:
This means that the LMIC has not completed it's activity. Usually means it's getting a downlink that requires more work (after it's sent your TX complete event). Wtih TTN, you have a 6 second (5+1) cycle, and there can be multiple downlink messages.  You'll get additional TX complete events as the LMIC finishes its work.  Really, what's needed in your app is a queue of application uplinks; when you get a TX complete, you can then send the next message (or you pull next message from queue). I've got a complete implementation of a queueing wrapper for the LMIC, but it's in C++ and it depends on other MCCI stuff. For no good reason, it's not on github on the moment. I'll try to get that repo published in the next week or so, and then there will at least be something for reference.

But this might be enough of a hint:  TXCOMPLETE does not mean LMIC idle; LMIC idle is controlled by the network, not really by you... a message cycle started by you with an uplink can go on for an arbitrarily long period (network sends something that needs response, LMIC responds, and that opens another RX window; and network again sends something....)

sw-dev-code, 2022-02-24:
@terrillmoore Thank you for your explanation. It seems like I missed the step where I need to wait for `os_queryTimeCriticalJobs()` to be false. 
Following issue helped me: https://github.com/manuelbl/ttn-esp32/issues/42

closed 2022-02-24 by sw-dev-code