mcci-catena/arduino-lmic #864: Is there a way to switch sx1276 to CW mode?

Miceuz, 2022-03-04:
I know it's not directly related to LMIC library, but I found no better place to ask on the internet. Please feel free to close if it's inappropriate.

Is there a way to make RFM95 (sx1276) transmit a continuous wave at 867.1 -  868.5MHz? Does LMIC support this? 

I can see `CFG_TxContinuousMode` define in `limic.h`, but I don't see CW on spectrum analyzer after enabling it.

Miceuz, 2022-03-04:
Sorry for the noise, wrong measurements

closed 2022-03-04 by Miceuz

terrillmoore, 2022-03-04:
For FCC/CE/etc. certification, it seems to me that it would be nice if this were both a conditional compile option (to enable the feature) and a run time option (so we don't have to reload firmware to do this test).  Often it's inconvenient for the RF lab to change firmware.