mcci-catena/arduino-lmic #873: Generating of random numbers doesn't terminate on Ideetron LoRa Nexus Board

tgesthuizen, 2022-05-15:
Hi there,
I tried out the Ideetron LoRa Nexus Board with the ttn-otaa script that is supplied by this library.
After editing the required configurations and uploading to the board I noticed that the microcontroller gets stuck in the startup code of lmic.
It turns out that a loop trying to obtain random numbers from a noise signal (I assume, at least) never terminates.
The following patch solves the issue for me by using another, though admittedly very bad, way of obtaining random numbers:
[https://gist.github.com/tgesthuizen/9f9041cc2e563a83c6b41b8b81da1fb8](0001-Replace-non-terminating-random-generator.patch)
It is just meant to pinpoint the issue.
The nicest fix that I could imagine would be a way to have a user supply his own random function for the library and keep the current one as the default as it seems to work for other users of the library.
For my project I already have a decent source of randomness and I don't like other ones being forced on me, especially if they get stuck :)

This is the first time I'm tinkering with LoRaWAN and issuing a problem here.
Because of this I didn't dare to mark this as a bug straight away, especially as the Nexus Board is on the list of supported devices and such an obvious bug seems unlikely to be missed.
Therefore I deem it not unlikely that I made a mistake, feedback is very welcome.

**Environment**

- LMIC version: 4.1.1, the one downloaded by the Arduino IDE
- Version of Arduino IDE being used: 1.8.19, the one shipped by Debian bookworm/sid
- Network provider: TTN
- Region: EU868
- Board: Ideetron LoRa Nexus Board
- Radio: HopeRF RFM95W

terrillmoore, 2022-05-16:
Sorry you're having problems. Not certain. I don't have a Nexus board to test with, and the "support" claims were inherited when I forked the library. I would suggest quickly trying the last version of the https://github.com/matthijskooijman/arduino-lmic, and see if that works any differently.  I've never seen the random number generator fail, so the problem is likely just an error that crept in during one of the many refactors of the code, having to do with how the code thinks the Nexus is wired up to the Arduino.

tgesthuizen, 2022-05-17:
I tried it with the old repo and the master branch fails for a different reason on the board, but initialization succeeds.
So I took the time to `git bisect` through the whole git history in order to find a culprit.
And thankfully git points fingers at a rather promising commit at last:
`~/src/arduino-lmic$ git bisect bad
e2d6f4851087d6b4ae0b4e2e55415ba7c57ad262 is the first bad commit
commit e2d6f4851087d6b4ae0b4e2e55415ba7c57ad262
Author: Terry Moore <tmm@mcci.com>
Date:   Sun May 10 18:01:44 2020 -0400

    Fix #524: simplify interrupt handling

 README.md         | 46 +++++++++++++++++++++++++++----------
 src/hal/hal.cpp   | 68 +++++++++++++++++++++++++++++++++++++++----------------
 src/lmic/hal.h    | 11 +++++++++
 src/lmic/lmic.h   |  4 ++--
 src/lmic/oslmic.c |  2 ++
 src/lmic/radio.c  | 21 ++++-------------
 6 files changed, 102 insertions(+), 50 deletions(-)`

I didn't look into the commit yet but it does touch the offending code section.