mcci-catena/arduino-lmic #874: ESP32 TTO Ok with ttn-otaa example and bad with esp32 RFM95W

ebelouet, 2022-05-16:
HI,
I using ESP32 TTO Ok with ttn-otaa example that running very well but with with esp32 RFM95W board, I have starting message and after nothing.
Do you knwo where is problem ?
Sincerely
Eric

terrillmoore, 2022-05-16:
Sorry you're having problems. It is probably the wiring. Without more information, it's hard to speculate.

ebelouet, 2022-05-16:
Yes, but this schema run very well with 3.3.3 libray.
[image]
// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 18,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 23,
    .dio = {26, 33, 32},
};

terrillmoore, 2022-05-17:
Perhaps you can use `git bisect` to find the error. Sorry, without hardware, I can't do much.

ebelouet, 2022-05-17:
HI,
Thanks, do you have a schematic with esp32 and RFM95W module to check with it?

terrillmoore, 2022-05-17:
Any TTGO or Heltec device seem to work well based on what others have told me. I know there are lots of users.

ebelouet, 2022-07-13:
HI,

closed 2022-07-18 by ebelouet