mcci-catena/arduino-lmic #876: Where can I find default values for some of LoRaWAN parameters library use?

coded-life, 2022-05-25:
For LoRaWAN certification process I need values for following parameters:

```
RECEIVE_DELAY1
RECEIVE_DELAY2
JOIN_ACCEPT_DELAY1
JOIN_ACCEPT_DELAY1
MAX_FCNT_GAP
ADR_ACK_LIMIT
ADR_ACK_DELAY
ACK_TIMEOUT
```

I can't find a values anywhere in library. 
Thank you in advance.

terrillmoore, 2022-05-26:
The values are all as specified in the 1.0.3 spec and the defaults in the regional parameters. (There is no way to configure them, and no plan to make this configurable.)

coded-life, 2022-05-26:
@terrillmoore Thank you. So at the question which asked if I'm using default values regarding the EU868 Regional parameters I can answer with YES?

terrillmoore, 2022-05-26:
Yes, that's what I believe. (Apart from bugs, of course.)

coded-life, 2022-05-28:
@terrillmoore Thanks, closing the issue.

closed 2022-05-28 by coded-life