mcci-catena/arduino-lmic #883: India Frequency Settings in LoRaBase

UdayChaduvula, 2022-07-02:
I have found that India lora frequency setting are missing in the LoRaBase.h file.

terrillmoore, 2022-07-02:
Sorry you're having trouble.

Can you be more specific? The LMIC works well in the India region, MCCI India uses it that way all the time....

Best regards,
--Terry

UdayChaduvula, 2022-07-04:
Hi,

I want to create a LoRaWAN node and it should be compatible with India lora
frequency (866 MHz). I am using the SX1272 module. Where can I enable India
Frequency in the code? In the configuration file I found only AU and US and
also in the LoRaBase file. Where can I enable LoRa frequency in the code?

dhineshkumarmcci, 2022-07-04:
Hi Uday,

> In the configuration file I found only AU and US and also in the LoRaBase file. 

- Can you please point out to the file and line number?
- Could be helpful if you can share the permalink (which redirects to particular line / set of lines in a file).

Please find the permalink for India band in `lorabase.h`
- https://github.com/mcci-catena/arduino-lmic/blob/master/src/lmic/lorabase.h#L377-L420

UdayChaduvula, 2022-07-04:
Hi Mcci-Catena/Arduino-Lmic,

Sorry for the issue, I have used the IBM LMIC framework library for my
project, I thought it was the same. Apologies from my side.

I have found one more issue in the in *lmic_project_config.h*, SX272 module
is missing in the file. Please find the paralink below.

https://github.com/mcci-catena/arduino-lmic/blob/8d378ea410887d3fb08ea2c9acce95cc4c047788/project_config/lmic_project_config.h#L9

UdayChaduvula, 2022-07-04:
Hi Mcci-Catena/Arduino-Lmic,

My LoRa board SX1272 MBED Shield is missing in the lmic_hal_boards.h file.
Please find the paralink below:
https://github.com/mcci-catena/arduino-lmic/blob/8d378ea410887d3fb08ea2c9acce95cc4c047788/src/arduino_lmic_hal_boards.h#L43

dhineshkumarmcci, 2022-07-04:
> Hi Mcci-Catena/Arduino-Lmic, Sorry for the issue, I have used the IBM LMIC framework library for my project, I thought it was the same. Apologies from my side. I have found one more issue in the in [...]

You can edit line 9 in the file `lmic_project_config.h` with sx1272 instead of sx1276.

> #define CFG_sx1272_radio 1

- https://github.com/mcci-catena/arduino-lmic/tree/master#selecting-the-target-radio-transceiver

UdayChaduvula, 2022-07-04:
Hi Dinesh,

I am awaiting your response.

UdayChaduvula, 2022-07-04:
> Please find the README link to add your board manually.

Thanks for the update Dhinesh Kumar,

How to generate DEV EUI, APP EUI and APP KEY ? Could you please support me
on this ?

UdayChaduvula, 2022-07-15:
> Hi Dhanush,
>
> I want to send the payload in ABP mode when the device receives the input
> in serial. How Can I do it? Could you help me in this regard?
>
> I am able to do it in the scheduled timer which is already present in the
> code. Please find the Paralink below
>
>
> https://github.com/mcci-catena/arduino-lmic/blob/8d378ea410887d3fb08ea2c9acce95cc4c047788/examples/ttn-abp/ttn-abp.ino#L140
>
> On Mon, Jul 4, 2022 at 6:49 PM Dhinesh Kumar Pitchai <

Hi Dhanush,

Greetings of the day!

How can I change the Spreading factor in the code?

dhineshkumarmcci, 2022-07-15:
Hi Uday, you cal use the below API to change SF.

https://github.com/mcci-catena/arduino-lmic/blob/8d378ea410887d3fb08ea2c9acce95cc4c047788/src/lmic/lmic.c#L2751-L2754

Please find the list of Spread Factor definitions over here

https://github.com/mcci-catena/arduino-lmic/blob/8d378ea410887d3fb08ea2c9acce95cc4c047788/src/lmic/lorabase.h#L409-L420

For example - `LMIC_setDrTxpow(DR_SF12, 30);`

UdayChaduvula, 2022-07-16:
Hi Dinesh,

How can I assign my Own DEV ADDR in the OTAA Code, I don't want to generate
it from the LoRaWan Gateway.

dhineshkumarmcci, 2022-07-18:
Hi Uday, please refer to the API below for more details:

https://github.com/mcci-catena/arduino-lmic/blob/4342770945ee2887c14969cec669861b387947ee/src/lmic/lmic.c#L3004-L3026

Also please do refer to the API description just above the definition.

UdayChaduvula, 2022-09-23:
Hi Mcci-Catena/Arduino-Lmic,

Greetings of the day!

How can I mention the Indian region  in OTAA code? Or the code
automatically identifies itself from the library?

UdayChaduvula, 2022-09-24:
Hi,

Can anyone please respond to the trailing mail ?

terrillmoore, 2022-09-25:
Please check the instructions [in the README](https://github.com/mcci-catena/arduino-lmic#selecting-the-lorawan-region-configuration).  

[image]